tinygo-org/tinygo #2735: WASI reactor mode

slinkydeveloper, 2022-03-24:
Hi all,

I've been experimenting with tinygo having some fun prototyping a go wasm application hosted by wasmtime-java. Unfortunately, I hit a bug/missing feature when trying to access the memory: https://github.com/kawamuray/wasmtime-java/issues/21. As described in that issue, tinygo cannot compile in [reactor mode](https://github.com/WebAssembly/WASI/blob/2ccb606926bee332a7b0a486e150d1053a493977/design/application-abi.md#current-unstable-abi), hence preventing wasmtime-java to access its memory. Although this might be recognized as a limitation of the engine (apparently others, at least from the documentation, can switch the usage mode independently from the exports https://wasmedge.org/book/en/dev/go.html#a-simple-function-run), I think it would be nice if tinygo can support the reactor mode.

I tried to prototype the support myself following what was done for clang https://reviews.llvm.org/D62922 and this is roughly what I needed to come up with a correct export:

* Patch manually [`wasi.json`](https://github.com/tinygo-org/tinygo/blob/release/targets/wasi.json#L11) adding the ldflag `"--entry=_initialize"`
* Rename the export `_start` to `_initialize`: https://github.com/tinygo-org/tinygo/blob/release/src/runtime/runtime_wasm_wasi.go#L15

If you want I can also provide a proper PR, but i need some guidance :)

codefromthecrypt, 2022-09-07:
There may be issues after this, but I guess the main question is how to control the `//export` name (ex based on flags). Is templating supported?

https://github.com/tinygo-org/tinygo/blob/release/src/runtime/runtime_wasm_wasi.go#L16

aykevl, 2024-09-01:
https://github.com/tinygo-org/tinygo/pull/4451 implements reactor mode for wasip1.

deadprogram, 2025-02-14:
Closing since this was implemented some time ago. Thanks everyone!

closed 2025-02-14 by deadprogram