tinygo-org/tinygo #2757: Missing runtime.memhash implementation

eliasnaur, 2022-04-07:
I'm getting a bit further in porting Gio to TinyGo. I've hit another missing symbol:

```
d.lld-11: error: undefined symbol: runtime.memhash
>>> referenced by maphash.go:234 (/nix/store/3v2l94h7pllq6za9km3388cyd5agrln7-go-1.17.7/share/go/src/hash/maphash/maphash.go:234)
>>>               /tmp/tinygo2131701156/main.o:(hash/maphash.rthash)
>>> referenced by maphash.go:235 (/nix/store/3v2l94h7pllq6za9km3388cyd5agrln7-go-1.17.7/share/go/src/hash/maphash/maphash.go:235)
>>>               /tmp/tinygo2131701156/main.o:(hash/maphash.rthash)
error: failed to link /tmp/tinygo2131701156/main: exit status 1

$ tinygo version
tinygo version 0.23.0-dev linux/amd64 (using go version go1.17.7 and LLVM version 11.1.0)
```

dkegel-fastly, 2022-04-07:
On Linux, we get this from a .s file in regular go, so https://github.com/tinygo-org/tinygo/pull/2688 will probably fix it if/when that lands.

dgryski, 2022-04-08:
I thought that was fixed by https://github.com/tinygo-org/tinygo/pull/2664 ?

dkegel-fastly, 2022-04-08:
Oh, maybe, yeah.  I probably got runtime confused with syscall.
Elias, can you point us to a branch we can build to reproduce the problem?

eliasnaur, 2022-04-08:
My apologies, I had my branches crossed and was on an early commit. The issue is fixed on `dev` tip.

closed 2022-04-08 by eliasnaur