tinygo-org/tinygo #2758: Tinygo error: requires go version 1.15 through 1.17, got go1.18

andyzyb, 2022-04-08:
$tinygo version: 

tinygo version 0.22.0 linux/amd64 (using go version go1.18 and LLVM version 13.0.0)

$ tinygo build  -o main.wasm main.go
error: requires go version 1.15 through 1.17, got go1.18

This seems strange why it only supports <= 1.17 but itself built with 1.18

dankegel, 2022-04-08:
We've been holding off applying e8737d8285aeaa868bb449c4e4a947646204af36 from 
https://github.com/tinygo-org/tinygo/pull/2725 because it fails a test on windows.
We probably shouldn't wait.  In the meantime, you can apply it yourself.

dankegel, 2022-04-09:
Try it now.

closed 2022-04-11 by dkegel-fastly

freecodewu, 2022-04-11:
[image]

I also have the issue.

freecodewu, 2022-04-11:
> [image]
> 
> I also have the issue.

macOS

dkegel-fastly, 2022-04-11:
The fix is in the dev branch and will be part of tinygo 0.23 when it is released.

If you want to try it now, it's not too hard to build the dev branch.

freecodewu, 2022-04-12:
> 

if I use go1.17 , will the issue exist。

dkegel-fastly, 2022-04-12:
tinygo 0.22 supports go 1.17.  Using go 1.17 for now is easiest.