tinygo-org/tinygo #2759: Type assertion to empty interface type crashes with `type assert failed`

eliasnaur, 2022-04-08:
Reproducer program,

```
package main

import "fmt"

type Empty interface {
}

func main() {
	var intf interface{} = new(int)
	typed := intf.(Empty)
	fmt.Println(typed)
}
```

crashes on my `stm32f469disco` target with:

`panic: runtime error: type assert failed`

Issue found while running (more) Gio code.

aykevl, 2022-04-10:
Did a quick test and this is also present in the release branch (so it isn't caused by #2753).

aykevl, 2022-04-10:
Simplified to avoid use of `fmt`:

```go
package main

type Empty interface {
}

func main() {
        var intf interface{} = 3
        typed := intf.(Empty)
        use(typed)
}

func use(typed Empty) {
        print("use: ")
        println(typed.(int))
}
```

It panics on the interface type assert in the main function.

inindev, 2023-01-17:
I ran into this same issue today using an esp32c3. My output is via uart1 and I am using a local build of the ```dev``` branch which was built at commit 33489d63. I see the same issue using ```dev``` commit 776dabb2. Let me know if it would be helpful to test against a different branch or commit.

The first call to ```fmt.Printf``` succeeds, then the second call results in: "panic: runtime error: type assert failed"

```
package main

import (
    "fmt"
    "time"
)

func main() {
    fmt.Printf("hello world 1\r\n")
    fmt.Printf("hello world 2\r\n")

    for {
        time.Sleep(time.Second)
    }
}
```

output
```
hello world 1
panic: runtime error: type assert failed
```