tinygo-org/tinygo #2760: Add support for Mac M1

supcik, 2022-04-09:
I just managed to compile and install tinygo on the new Mac M1 inside a Docker container. This allows me to use tinygo on the M1 while waiting for a native version.

I think that it is not so hard to extend the _official_ Docker image "tinygo/tinygo" for arm64 as well (multi-architecture image).

A small issue is the [extraction of `xtensa-esp32-elf-ld`](https://github.com/tinygo-org/tinygo/blob/3883550c446fc8b14115366284316290f3d3c4b1/Dockerfile#L25).

If `xtensa-esp32-elf-ld` is really needed (can someone confirm ?), then it is possible to compile it using the instructions from [the espressif web site](https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/get-started/linux-setup-scratch.html)

_Note 1: I just had to use the default branch instead of the `esp-2019r2` tag described on the web page_.
_Note 2: The tool can't be compiled as root, and we have to handle this in the Dockerfile_

So, if `xtensa-esp32-elf-ld` is required, it is possible to compile it instead of extracting from a tar file. It will take more time, but it allows getting a native version of the linker.

I would be happy to work on this issue if you think that it makes sense.

deadprogram, 2022-04-11:
> So, if xtensa-esp32-elf-ld is required, it is possible to compile it instead of extracting from a tar file. It will take more time, but it allows getting a native version of the linker.

This is a requirement for compiling for esp8266 and esp32 targets.

aykevl, 2022-04-11:
I've been working on replacing `xtensa-esp32-elf-ld` with native ld.lld, but I need to finish it some day: https://github.com/espressif/llvm-project/pull/53 (it almost works but still has some issues with some corner cases).
Might be worth including that patch in our fork of LLVM though to avoid the `xtensa-esp32-elf-ld` dependency.

aykevl, 2022-04-11:
I think that TinyGo should also work with Rosetta 2 emulation. Or are there any problems with it?

supcik, 2022-04-18:
Thank you for your reply. Yes, it does indeed work with Rosetta 2, but I am a big fan of [devcontainers](https://code.visualstudio.com/docs/remote/containers) and I wanted to use them for Tinygo as well.

It was also a good way to have a standard containerized developer environment for the contributors of TinyGo.

I tried to add a step in the Github action to build the Xtensa Tools from the source files, but it took more than the allowed 6 hours to complete 😞. 

Let me know if you think that a Docker Image for ARM64 would be a good contribution or not. If yes, I will find ways to produce it. If no, then feel free to close this issue (I won't be sad 😄).

arbourd, 2022-04-23:
I think it would be a great contribution @supcik! 

Do you have that PR/build that indicates the 6h build for xtensa tools?

supcik, 2022-04-25:
This is the result of the build : https://github.com/supcik/tinygo/runs/5979455281?check_suite_focus=true

I did not find how to get rid of the spinning wheel, so the log is very long, and the job aborted after 6 hours.

arbourd, 2022-04-27:
did you try building this from source locally @supcik?

arbourd, 2022-04-27:
So I don't think we'll ever be able to beat the 6h limit set by Actions. At the moment we're building TinyGo (and LLVM) ourselves from scratch and not even worrying about xtensa and it takes longer than 6h because of QEMU.

We either need native arm64 Actions VMs or not building LLVM as part of this process.

deadprogram, 2022-04-28:
Hopefully in the next iterations I will be able to get around to building LLVM as a separate container to avoid the recompilation when LLVM itself it not being changed.

supcik, 2022-04-29:
> did you try building this from source locally @supcik?

Yes, I did... and it worked. I was able to compile the tools on my M1:

```
$ file xtensa-esp32-elf-ld
xtensa-esp32-elf-ld: ELF 64-bit LSB pie executable, ARM aarch64, version 1 (SYSV),
  dynamically linked, interpreter /lib/ld-linux-aarch64.so.1,
  BuildID[sha1]=55e68880bb7bfed620ce3f52038dbbddf77e8374, for GNU/Linux 3.7.0, stripped
```

arbourd, 2022-05-04:
I think we will need some custom hardware to run arm64 builds. I don't know if this project has access to any, but we can build everything in 1-2h with 32 or 64 core machines. Thankfully, the LLVM build takes advantage of N cores.

arbourd, 2022-05-04:
You can definitely cross compile on an `amd64` Mac for an `arm64` target. I've done it with `ripgrep`. 

That being said -- a big part of the conversation ☝🏻 is about getting that arm64 build in a container (which would be linux/arm64) which benefits both the Linux and Darwin crews.

aykevl, 2022-05-04:
Oh, in that case it's probably quite easy. I didn't realize this would be a _Linux_ Docker image (not building a macOS image) - I'm not very familiar with Docker. You can basically build a Linux binary like in #2633 but with some small modifications. Specifically:

* Modify the Makefile to add support for aarch64
* Build LLVM using `make llvm-build CROSS=aarch64-linux-gnu`
* Build TinyGo using `make CROSS=aarch64-linux-gnu`
* Optionally build `wasm-opt` using `make binaryen CROSS=aarch64-linux-gnu`

You can then replace the `bin/tinygo` binary in an existing release tarball and build a Docker image with the result.

Here is an untested patch to the Makefile to make this possible:

```diff
diff --git a/Makefile b/Makefile
index f05dde62..c590accb 100644
--- a/Makefile
+++ b/Makefile
@@ -68,6 +68,15 @@ ifneq ($(CROSS),)
             -DLLVM_TARGET_ARCH=ARM
         GOENVFLAGS = GOARCH=arm CC=$(CC) CXX=$(CXX) CGO_ENABLED=1
         BINARYEN_OPTION += -DCMAKE_C_COMPILER=$(CC) -DCMAKE_CXX_COMPILER=$(CXX)
+    else ifeq ($(CROSS), aarch64-linux-gnu)
+        # Assume we're building on a Debian-like distro, with QEMU installed.
+        LLVM_CONFIG_PREFIX = qemu-aarch64 -L /usr/aarch64-linux-gnu/
+        # The CMAKE_SYSTEM_NAME flag triggers cross compilation mode.
+        LLVM_OPTION += \
+            -DCMAKE_SYSTEM_NAME=Linux \
+            -DLLVM_TARGET_ARCH=AArch64
+        GOENVFLAGS = GOARCH=arm64 CC=$(CC) CXX=$(CXX) CGO_ENABLED=1
+        BINARYEN_OPTION += -DCMAKE_C_COMPILER=$(CC) -DCMAKE_CXX_COMPILER=$(CXX)
     else
         $(error Unknown cross compilation target: $(CROSS))
     endif
```

aykevl, 2022-05-04:
> You can definitely cross compile on an `amd64` Mac for an `arm64` target. I've done it with `ripgrep`.

Yes you can, but in the case of compilers this is usually a lot harder than with most programs. For example, in TinyGo we need to run a binary (built to the target architecture) to be able to get the correct linker flags for LLVM.

arbourd, 2022-05-04:
> Oh, in that case it's probably quite easy. I didn't realize this would be a _Linux_ Docker image (not building a macOS image) - I'm not very familiar with Docker. You can basically build a Linux [...]

This is great! We actually build tinygo for `linux/arm64` with the current Makefile (excluding xtensa support) and it works just great with the current Makefile. The way Docker does it, is wrapping everything with qemu (which I assume is what cross is doing). The only problem is the build time for LLVM. On a 2 core machine (the default for Actions) you would be looking at 12-24h. 

I'd like to help get an official arm64 Docker image. This benefits Mac users because, under the hood, Docker runs on a arm64 linux VM -- but also anyone using arm64 Linux devices 😁

We're using a 56 core machine to build tinygo in about 1.5h. Here's the Dockerfile: https://github.com/suborbital/subo/blob/bd0604ee1cc5ed56fe7d82a598b06102dc33afda/builder/docker/tinygo/Dockerfile.base

supcik, 2022-05-04:
What I did was to take the "official" tinygo Dockerfile (https://github.com/tinygo-org/tinygo/blob/release/Dockerfile) and to replace the X86 specific part :

```Dockerfile
# tinygo-xtensa stage installs tools needed for ESP32
FROM tinygo-llvm-build AS tinygo-xtensa

ARG xtensa_version="1.22.0-80-g6c4433a-5.2.0"
RUN cd /tmp/ && \
    wget -q https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-${xtensa_version}.tar.gz && \
    tar xzf xtensa-esp32-elf-linux64-${xtensa_version}.tar.gz && \
    cp ./xtensa-esp32-elf/bin/xtensa-esp32-elf-ld /usr/local/bin/ && \
    rm -rf /tmp/xtensa*
```

The code above downloads and extracts an x86 version of the xtensa tools. Instead, I made a Dockerfile that rebuilds those tools from the source. The result is here : https://github.com/supcik/tinygo/blob/docker-arm64/Dockerfile

It builds on my Mac M1 and I have a local Docker image with a complete tinygo (incl ESP32 support)

I could upload this image to Dockerhub, but I would prefer to build the image using Github Actions. This takes however too long because Github actions builds arm64 images using qemu emulation.

arbourd, 2022-05-04:
> This takes however too long because Github actions builds arm64 images using qemu emulation.

Exactly. This is awesome tho @supcik. 

@deadprogram, @aykevl do you have any donated or usable hardware that we could use for custom runners? In our case, spin up giant machines to get it done quickly. If using AWS, we could probably use both Graviton and AMD64 chipsets in tandem to make it quick.

aykevl, 2022-05-05:
But I don't think you need arm64 hardware to build the container? As I described before.
We have in fact multiple Docker containers:
 - one is the dev container (tinygo-dev), that's large but easier to debug
 - the other is the release container [here](https://github.com/tinygo-org/docker/blob/master/Dockerfile) that basically just downloads the latest stable release and extracts it inside the container

What I would suggest is to cross-compile a release tarball using the method [I described above](https://github.com/tinygo-org/tinygo/issues/2760#issuecomment-1117820212) (I'm sure this can be done on GitHub Actions in an amd64 container) and then extract the result in the same way in a Docker container.

aykevl, 2022-05-05:
In other news, CircleCI is finally giving some signs of adding arm64 support! https://circleci.canny.io/cloud-feature-requests/p/support-new-m1-arm-based-macs
We can't really use CircleCI because of the 1 hour build time limit (LLVM takes longer than that) but hopefully other CI providers will follow.

aykevl, 2022-06-23:
Based on https://github.com/actions/virtual-environments/issues/2187#issuecomment-1164711613, we might be getting really close to an official M1 runner for GitHub Actions!

aykevl, 2022-11-05:
Is there anything else we need to do for this bug? We currently do support darwin/arm64 and we have a linux/arm64 release. We don't have a darwin/arm64 release at the moment, because our CI provider (GitHub Actions) doesn't support it yet.

supcik, 2022-11-05:
OK. Thank you. Then I guess that you can close it.

arbourd, 2022-11-07:
Ya from a Docker perspective this done. Thanks eveyone.

dkegel-fastly, 2023-03-25:
Closing, as m1 support seems to work fairly well.

closed 2023-03-25 by dkegel-fastly