tinygo-org/tinygo #2762: On windows, compiling test compress/flate fails in interp allocating 802KB?

dkegel-fastly, 2022-04-10:
CI fails on windows (and not when cross-compiling!) with go 1.18.
Here's a reconstructed log from https://github.com/tinygo-org/tinygo/pull/2761's https://github.com/tinygo-org/tinygo/runs/5958440739 :

```
$ tinygo test compress/flate
...
ok  	compress/bzip2	0.400s
runtime: VirtualAlloc of 802816 bytes failed with errno=1455
fatal error: out of memory

runtime stack:
runtime.throw({0x48ee007?, 0xc0d47b2000?})
	C:/hostedtoolcache/windows/go/1.18.0/x64/src/runtime/panic.go:992 +0x76
runtime.sysUsed(0xc0d4748000, 0xc4000)
	C:/hostedtoolcache/windows/go/1.18.0/x64/src/runtime/mem_windows.go:83 +0x1c9
runtime.(*mheap).allocSpan(0x66183c0, 0x62, 0x0, 0x1)
	C:/hostedtoolcache/windows/go/1.18.0/x64/src/runtime/mheap.go:1279 +0x428
runtime.(*mheap).alloc.func1()
	C:/hostedtoolcache/windows/go/1.18.0/x64/src/runtime/mheap.go:912 +0x65
runtime.systemstack()
	C:/hostedtoolcache/windows/go/1.18.0/x64/src/runtime/asm_amd64.s:469 +0x4e

goroutine 3269 [running]:
runtime.systemstack_switch()
	C:/hostedtoolcache/windows/go/1.18.0/x64/src/runtime/asm_amd64.s:436 fp=0xc008275c88 sp=0xc008275c80 pc=0x462fc0
runtime.(*mheap).alloc(0xc4000?, 0x62?, 0x0?)
	C:/hostedtoolcache/windows/go/1.18.0/x64/src/runtime/mheap.go:906 +0x65 fp=0xc008275cd0 sp=0xc008275c88 pc=0x427e45
runtime.(*mcache).allocLarge(0x0?, 0xc3500, 0x1)
	C:/hostedtoolcache/windows/go/1.18.0/x64/src/runtime/mcache.go:213 +0x85 fp=0xc008275d20 sp=0xc008275cd0 pc=0x4180e5
runtime.mallocgc(0xc3500, 0x487db20, 0x1)
	C:/hostedtoolcache/windows/go/1.18.0/x64/src/runtime/malloc.go:1096 +0x5a5 fp=0xc008275d98 sp=0xc008275d20 pc=0x40e545
runtime.makeslice(0x186a0?, 0x0?, 0x0?)
	C:/hostedtoolcache/windows/go/1.18.0/x64/src/runtime/slice.go:103 +0x52 fp=0xc008275dc0 sp=0xc008275d98 pc=0x44c212
github.com/tinygo-org/tinygo/interp.rawValue.clone(...)
	D:/a/tinygo/tinygo/interp/memory.go:679
github.com/tinygo-org/tinygo/interp.(*rawValue).clone(0x0?)
	<autogenerated>:1 +0x4e fp=0xc008275e18 sp=0xc008275dc0 pc=0x7126ee
github.com/tinygo-org/tinygo/interp.object.clone(...)
	D:/a/tinygo/tinygo/interp/memory.go:52
github.com/tinygo-org/tinygo/interp.(*memoryView).store(0xc009a5e7b0, {0x497d038, 0xc005048ef0}, {0xc00810e000?})
	D:/a/tinygo/tinygo/interp/memory.go:316 +0x1f5 fp=0xc008275f78 sp=0xc008275e18 pc=0x70d055
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc00[51](https://github.com/tinygo-org/tinygo/runs/5958440739?check_suite_focus=true#step:18:51)[53](https://github.com/tinygo-org/tinygo/runs/5958440739?check_suite_focus=true#step:18:53)900, 0xc002d2f9a0, {0xc007631510, 0x1, 0x0?}, 0xc009958a80, {0xc008107a50, 0x8})
	D:/a/tinygo/tinygo/interp/interpreter.go:[57](https://github.com/tinygo-org/tinygo/runs/5958440739?check_suite_focus=true#step:18:57)6 +0x2a16 fp=0xc008276ba0 sp=0xc008275f78 pc=0x[70](https://github.com/tinygo-org/tinygo/runs/5958440739?check_suite_focus=true#step:18:70)2cf6
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc005153900, 0xc002d2f[72](https://github.com/tinygo-org/tinygo/runs/5958440739?check_suite_focus=true#step:18:72)0, {0x0, 0x0, 0x24?}, 0x0, {0x48e8a5e, 0x4})
```

deadprogram, 2022-04-11:
So this is basically the main blocker for a new TinyGo release with support for Go 1.18, correct?

Spidy104, 2022-04-15:
Sorry an offbeat question, is tinygo good for developing various kinds of embedded or iot stuff, or still does it lack something, no offense to the community or the creators, it is a wonderful project...

dkegel-fastly, 2022-04-15:
It's rather good.  The only annoying bits are 1) serialization of structs is a little harder, because 2) reflection isn't fully supported.  So if you need a platform that regular Go doesn't support, tinygo is more than good enough usually.

Spidy104, 2022-04-21:
> It's rather good. The only annoying bits are 1) serialization of structs is a little harder, because 2) reflection isn't fully supported. So if you need a platform that regular Go doesn't support, tinygo is more than good enough usually.

Is it good for developing professional IoT devices?

deadprogram, 2022-04-21:
> Is it good for developing professional IoT devices?

There are a number of us who think so! :smile_cat:

deadprogram, 2022-04-21:
Please feel to join our Slack community here: https://app.slack.com/client/T029RQSE6/CDJD3SUP6

aykevl, 2022-04-21:
Take a look here for how to get into the Slack workspace: https://github.com/tinygo-org/tinygo/#getting-help

aykevl, 2022-04-22:
I've almost reproduced this issue:

```
Ayke@sputnik-win10 MINGW64 ~/go/src/github.com/tinygo-org/tinygo (llvm14)
$ build/tinygo test compress/flate
Assertion failed: PrettyStackTraceHead == this && "Pretty stack trace entry destruction is out of order", file C:/Users/Ayke/go/src/github.com/tinygo-org/tinygo/llvm-project/llvm/lib/Support/PrettyStackTrace.cpp, line 229
```

...only I forgot that there's this one bug.

aykevl, 2022-04-22:
Seems to be working here:

```
$ build/tinygo test compress/flate
ld.lld: warning: duplicate /export option: hypot
ld.lld: warning: duplicate /export option: nextafter
ok      compress/flate  13.068s

$ build/tinygo version
tinygo version 0.23.0-dev-24f2bb6c windows/amd64 (using go version go1.18.1 and LLVM version 14.0.0)
```

This is with LLVM 14 because I didn't want to do another LLVM build. It might be that it doesn't run out of memory with LLVM 14. Or it might be that it just works on my laptop and not in other places. (Although I looked at the memory consumption and `tinygo.exe` didn't seem to use an exceptionally large amount of RAM).

deadprogram, 2022-04-22:
@dkegel-fastly is there a straightforward way to test this branch against the Windows CI build that is failing?

dkegel-fastly, 2022-04-22:
yes, left a note on that PR.

dankegel, 2022-04-23:
Reproduces reliably on a fresh windows installation with clang 14 in wasi like so:
```
$ build/tinygo clean
$ build/tinygo test -target wasi compress/flate
panic: runtime error: slice out of range
Error: failed to run main module `C:\Users\dank\AppData\Local\Temp\tinygo3559047210\main`

Caused by:
    0: failed to invoke command default
    1: wasm trap: wasm `unreachable` instruction executed
       wasm backtrace:
           0: 0x977b - runtime.abort
                           at C:\Users\dank\src\tinygo\src\runtime\runtime_tinygowasm.go:62:6
                     - runtime.runtimePanic
                           at C:\Users\dank\src\tinygo\src\runtime\panic.go:20:7
           1: 0x8d12 - runtime.slicePanic
                           at C:\Users\dank\src\tinygo\src\runtime\panic.go:42:14
           2: 0x9141e - compress/flate.TestWriterPersistentError
                           at C:\Program Files\Go\src\compress\flate\deflate_test.go:692:7
           3: 0x5510b - testing.tRunner
                           at C:\Users\dank\src\tinygo\src\testing\testing.go:366:4
           4: 0x54d82 - (*testing.T).Run

           5: 0x55e29 - testing.runTests$1
                           at C:\Users\dank\src\tinygo\src\testing\testing.go:469:9
           6: 0x5510b - testing.tRunner
                           at C:\Users\dank\src\tinygo\src\testing\testing.go:366:4
           7: 0x2b90b - testing.runTests
                           at C:\Users\dank\src\tinygo\src\testing\testing.go:467:9
                     - (*testing.M).Run
                           at C:\Users\dank\src\tinygo\src\testing\testing.go:443:29
           8: 0x27573 - main.main
                           at C:\Users\dank\AppData\Local\go-build\e3\e34211c3d896df81854d4008cd958f89031c9c35c5bb34707a1dfa2ced88d631-d:127:15
                     - runtime.run$1
                           at C:\Users\dank\src\tinygo\src\runtime\scheduler_any.go:25:11
           9: 0x26483 - <goroutine wrapper>
                           at C:\Users\dank\src\tinygo\src\runtime\scheduler_any.go:23:2
          10: 0xae0fd - <unknown>!tinygo_launch
          11: 0x2634f - (*internal/task.Task).Resume
                           at C:\Users\dank\src\tinygo\src\internal\task\task_asyncify.go:110:17
                     - runtime.scheduler
                           at C:\Users\dank\src\tinygo\src\runtime\scheduler.go:167:11
                     - runtime.run
                           at C:\Users\dank\src\tinygo\src\runtime\scheduler_any.go:28:11
                     - _start
                           at C:\Users\dank\src\tinygo\src\runtime\runtime_wasm_wasi.go:21:5

FAIL    compress/flate  45.567s
FAIL

```

aykevl, 2022-04-23:
Huh, I didn't have this issue.

Sadly my Windows install has suddenly given up on me after an update so I won't be able to test this soon.

dkegel-fastly, 2022-04-25:
Removing milestone since this seems to not be reproducible and/or needs more investigation.

aykevl, 2022-09-02:
This appears to be fixed now. Not sure what changed, maybe the move to LLVM 14?
I'm going to close this, unless someone can still reproduce this in the latest release.

Here I've added compress/flate to the Windows tests: https://github.com/tinygo-org/tinygo/pull/3125

closed 2022-09-02 by aykevl