tinygo-org/tinygo #2764: will there be ci support in jenkins plugin?

Neil-Boyle, 2022-04-11:
will there be **ci support in jenkins plugin**?

it seems there is **rare** cicd supports in tinygo projects.

any help will be welcome!   TIA

dkegel-fastly, 2022-04-11:
You mean something like https://plugins.jenkins.io/golang/, but for tinygo?  That sounds like a nice idea, maybe you could ask its maintainer, @orrc, about it; we'd probably need his help.

dkegel-fastly, 2022-04-11:
Until then, you could do it in a Dockerfile, e.g.

```
ARG UBUNTU=focal
FROM ubuntu:$UBUNTU as base
# Must redeclare args after each FROM, but don't have to repeat default value
ARG UBUNTU=focal
ENV DEBIAN_FRONTEND=noninteractive
RUN apt-get -qq update \
  && apt-get -qy --no-install-recommends install \
        build-essential \
        ca-certificates \
        cmake \
        curl \
        git \
        gnupg \
        libssl-dev \
        net-tools \
        ninja-build \
        pkg-config \
        software-properties-common \
        openssh-client \
        time \
        wget \
  && echo "Ubuntu basic tools install done"

# Install go
# Borrowed from https://github.com/tinygo-org/docker/blob/master/Dockerfile
ENV GO_RELEASE=1.17
ENV PATH=${PATH}:/usr/local/go/bin
RUN wget https://dl.google.com/go/go${GO_RELEASE}.linux-amd64.tar.gz && \
    tar xf go${GO_RELEASE}.linux-amd64.tar.gz -C /usr/local && \
    rm go${GO_RELEASE}.linux-amd64.tar.gz && \
    find /usr/local/go -mindepth 1 -maxdepth 1 ! -name 'src' ! -name 'VERSION' ! -name 'bin' ! -name 'pkg' -exec rm -rf {} + \
  && echo "Go installed"

# Install tinygo
ENV TINYGO_RELEASE=0.22.0
ENV PATH=${PATH}:/usr/local/tinygo/bin
RUN wget https://github.com/tinygo-org/tinygo/releases/download/v${TINYGO_RELEASE}/tinygo${TINYGO_RELEASE}.linux-amd64.tar.gz && \
    tar xf tinygo${TINYGO_RELEASE}.linux-amd64.tar.gz -C /usr/local && \
    rm tinygo${TINYGO_RELEASE}.linux-amd64.tar.gz \
  && echo "Tinygo release ${TINYGO_RELEASE} installed"
```

or, if you prefer alpine / want to build tinygo from source:

```
FROM golang:1.17.8-alpine3.15 as tinygo-base
ARG LLVM=12
# Setup container and install git
RUN apk update
RUN apk add git

# TODO: change this to v0.23.0 once that is released
# Until then, use commit where https://github.com/tinygo-org/tinygo/pull/2675 landed
ENV TINYGO_BRANCH=c55c996c5df01ea33cfe62bcc74943389761b3fd
RUN git clone https://github.com/tinygo-org/tinygo /tinygo && \
  cd /tinygo && \
  git fetch origin ${TINYGO_BRANCH} && \
  git reset --hard FETCH_HEAD && \
  git submodule update --init --recursive

# Install TinyGo dependencies
RUN apk --no-cache add \
  autoconf \
  automake \
  cmake \
  g++ \
  libffi-dev \
  make \
  musl-dev \
  llvm \
  clang \
  lld \
  clang-dev \
  llvm$LLVM-dev \
  python3 \
  samurai

# Setup some required ENV variables
ENV C_INCLUDE_PATH=/usr/include/llvm$LLVM
ENV CPLUS_INCLUDE_PATH=/usr/include/llvm$LLVM

# Compile TinyGo
RUN cd /tinygo/ && \
  go install -tags llvm$LLVM /tinygo/

# Compile WASM extensions to TinyGo
RUN cd /tinygo/ && \
  make binaryen && \
  make wasi-libc
```

orrc, 2022-04-11:
Yeah, I'd need to know what "CI support in Jenkins" looks like for you.

If you are thinking about the [Jenkins Go plugin](https://plugins.jenkins.io/golang/), then it should be fairly straightforward to include TinyGo releases in the list of versions that can be installed during a Jenkins build.

It would probably just require updating the [crawler](https://github.com/jenkins-infra/crawler/blob/master/golang.groovy) that runs daily, which updates the [list of Go versions](https://updates.jenkins.io/updates/org.jenkinsci.plugins.golang.GolangInstaller.json) that Jenkins displays on the tool installer settings page.

dkegel-fastly, 2022-04-11:
I think it's "have the jenkins go plugin install the given go version, and optionally with the given tinygo version", since tinygo needs a version of go to be installed.
  
The tinygo versions can be crawled from https://github.com/tinygo-org/tinygo/releases/

The versions of go that each version of tinygo supports could be scraped fairly easily from the source tarball there by awking the file builder/config.go for the string "requires go version %s through %s".

Neil-Boyle, 2022-04-12:
"have the jenkins go plugin install the given go version, and optionally with the given tinygo version"

Neil-Boyle, 2022-04-12:
----

Neil-Boyle, 2022-04-12:
TIA.

"have the jenkins go plugin install the given go version, and optionally with the given tinygo version"
--- 
yes its. i'm not going to do ci for tinygo, but for programs writen with tinygo, such as wasm.

===

Dockerfile can be replaced by existing tinygo/tinygo:0.22.0 image, as it can build wasm file for me

codefromthecrypt, 2022-09-07:
So, I think there's two paths for a formal plugin

* add TinyGo support to https://github.com/jenkinsci/golang-plugin
* create and register a new tinygo-plugin (in the jenkinsci org)

Either way, I think this issue would close out once a path is chosen, as the side effects would be in another org.

@Neil-Boyle end-user requests are always the best place to start. Can you file an issue to https://github.com/jenkinsci/golang-plugin to add TinyGo support to the existing plugin? If that's accepted, then we've a way out that doesn't involve copying some of it.

deadprogram, 2023-05-23:
Closing since this appears to be tasks for some other projects to incorporate. Thanks everyone!

closed 2023-05-23 by deadprogram