tinygo-org/tinygo #2771: misc hashmap cleanups

dgryski, 2022-04-12:
A reminder to myself to extract some of the common calculations to small helper functions and add `//go:nobounds` to avoid the extra nil-pointer checks.

dgryski, 2023-03-23:
Also clear out buckets when deleting keys to avoid pinning allocations.

Ancient patch rescued from `git stash` for some of the above cleanups.

[hashmap-extract-functions.patch](https://github.com/tinygo-org/tinygo/files/11047205/hashmap-extract-functions.patch)

dgryski, 2023-03-26:
Also size hinting should pre-allocate enough buckets so that we can actually insert `n` items without allocating.  Note that because of the logic in `hashmapShouldGrow` we need to *over* allocate to take into account the load factor calculation.   Move that calculation out to its own routine so that we don't need to duplicate the logic.

dgryski, 2023-03-27:
https://github.com/tinygo-org/tinygo/pull/3605

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing this issue. Thanks!