tinygo-org/tinygo #2773: docker run build wasm with go mod exception

Neil-Boyle, 2022-04-13:
in other way:
`tinygo build -o ./build/xxx.wasm -scheduler=none -target=wasi ./main.go
`
runs correct, with wasm file **created**.

docker run build wasm with go mod exception

i did run this command **in project root path, with go.mod file exist**:  
`docker run --rm -v $(pwd):/src tinygo/tinygo:0.22.0 tinygo build -o ./build/xxx.wasm -scheduler=none -target=wasi ./main.go
`

[image]

exception:
`go: go.mod file not found in current directory or any parent directory; see 'go help modules'
`

docker version:
```
Client:
 Cloud integration: 1.0.17
 Version:           20.10.8
 API version:       1.41
 Go version:        go1.16.6
 Git commit:        3967b7d
 Built:             Fri Jul 30 19:55:20 2021
 OS/Arch:           darwin/amd64
 Context:           default
 Experimental:      true

Server: Docker Engine - Community
 Engine:
  Version:          20.10.8
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.16.6
  Git commit:       75249d8
  Built:            Fri Jul 30 19:52:31 2021
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.4.9
  GitCommit:        e25210fe30a0a703442421b0f60afac609f950a3
 runc:
  Version:          1.0.1
  GitCommit:        v1.0.1-0-g4144b63
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```
local tinygo version:
`tinygo version 0.22.0 darwin/amd64 (using go version go1.17.2 and LLVM version 13.0.0)
`

any help will be welcome, TIA

Neil-Boyle, 2022-04-13:
resolved by customize tinygo dockerfile

closed 2022-04-14 by Neil-Boyle

mthienpont, 2025-04-24:
Hey, do you remember how you fixed this? having a similar issue in big 2025