tinygo-org/tinygo #2777: SIGSEGV in the compiler in transform.OptimizeAllocs call tree

ALTree, 2022-04-14:
```
$ tinygo version
tinygo version 0.22.0 linux/amd64 (using go version go1.17.8 and LLVM version 13.0.0)
```

The following program:

```
package main

var P *int

func f(p []int, m map[int]int) int {
	m[p[0]] = *P
	return 0
}

func main() {
	var m map[int]int
	f(make([]int, 1/f(nil, m)), m)
}
```

Occasionally triggers a panic in the compiler with stack trace:

```
$ tinygo build -o crash.o crash.go

fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x615 pc=0x4d3b3da]

runtime stack:
runtime.throw({0x50c8043, 0x7f6c4446a5c0})
	/opt/hostedtoolcache/go/1.17.6/x64/src/runtime/panic.go:1198 +0x71
runtime.sigpanic()
	/opt/hostedtoolcache/go/1.17.6/x64/src/runtime/signal_unix.go:719 +0x396

goroutine 53 [syscall]:
runtime.cgocall(0xe2ac70, 0xc0006e77f8)
	/opt/hostedtoolcache/go/1.17.6/x64/src/runtime/cgocall.go:156 +0x5c fp=0xc0006e77d0 sp=0xc0006e7798 pc=0xadc57c
tinygo.org/x/go-llvm._Cfunc_LLVMConstIntGetZExtValue(0x7f6c44011840)
	_cgo_gotypes.go:3365 +0x4d fp=0xc0006e77f8 sp=0xc0006e77d0 pc=0xccfccd
tinygo.org/x/go-llvm.Value.ZExtValue.func1({0x0})
	/home/runner/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20220121152956-4fa2ab2718f3/ir.go:854 +0x3f fp=0xc0006e7830 sp=0xc0006e77f8 pc=0xce761f
tinygo.org/x/go-llvm.Value.ZExtValue({0xc000464f50})
	/home/runner/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20220121152956-4fa2ab2718f3/ir.go:854 +0x19 fp=0xc0006e7848 sp=0xc0006e7830 pc=0xce75b9
github.com/tinygo-org/tinygo/transform.OptimizeAllocs({0x7f6c442aa320}, 0x0, 0xaf1a85)
	/home/runner/work/tinygo/tinygo/transform/allocs.go:51 +0x1ec fp=0xc0006e79d0 sp=0xc0006e7848 pc=0xdc662c
github.com/tinygo-org/tinygo/transform.Optimize({0xc0006ddde8}, 0xc0000fbce0, 0x2, 0x0, 0x5)
	/home/runner/work/tinygo/tinygo/transform/optimizer.go:70 +0x2c5 fp=0xc0006e7db0 sp=0xc0006e79d0 pc=0xdce5a5
github.com/tinygo-org/tinygo/builder.optimizeProgram({0x7f6c44051448}, 0xc0000fbce0)
	/home/runner/work/tinygo/tinygo/builder/build.go:867 +0x1f4 fp=0xc0006e7e10 sp=0xc0006e7db0 pc=0xde8fd4
github.com/tinygo-org/tinygo/builder.Build.func2(0x0)
	/home/runner/work/tinygo/tinygo/builder/build.go:476 +0x625 fp=0xc0006e7f70 sp=0xc0006e7e10 pc=0xde7985
github.com/tinygo-org/tinygo/builder.runJob(0xc00072b2c0, 0x0)
	/home/runner/work/tinygo/tinygo/builder/jobs.go:222 +0x4f fp=0xc0006e7fc0 sp=0xc0006e7f70 pc=0xdf1b0f
github.com/tinygo-org/tinygo/builder.runJobs·dwrap·13()
	/home/runner/work/tinygo/tinygo/builder/jobs.go:123 +0x2a fp=0xc0006e7fe0 sp=0xc0006e7fc0 pc=0xdf144a
runtime.goexit()
	/opt/hostedtoolcache/go/1.17.6/x64/src/runtime/asm_amd64.s:1581 +0x1 fp=0xc0006e7fe8 sp=0xc0006e7fe0 pc=0xb3c441
created by github.com/tinygo-org/tinygo/builder.runJobs
	/home/runner/work/tinygo/tinygo/builder/jobs.go:123 +0x5f8

goroutine 1 [chan receive]:
github.com/tinygo-org/tinygo/builder.runJobs(0x505c0a0, 0x50b87e1)
	/home/runner/work/tinygo/tinygo/builder/jobs.go:132 +0x625
github.com/tinygo-org/tinygo/builder.Build({0x7fff0f76150c, 0x8}, {0x7fff0f761504, 0x60}, 0xc0000fbce0, 0xc0002457e0)
	/home/runner/work/tinygo/tinygo/builder/build.go:495 +0x1d85
main.Build({0x7fff0f76150c, 0x8}, {0x7fff0f761504, 0x7}, 0x1)
	/home/runner/work/tinygo/tinygo/main.go:151 +0x8f
main.main()
	/home/runner/work/tinygo/tinygo/main.go:1337 +0x3aed
```

dgryski, 2022-04-14:
This code dereferences a nil pointer, writes into a nil map, accesses an out-of-bounds array element, and divides by zero.

aykevl, 2022-04-14:
Still, it shouldn't crash at _compile time_.

aykevl, 2022-04-14:
@ALTree does this test case come out of a fuzzer?

aykevl, 2022-04-14:
Fix: https://github.com/tinygo-org/tinygo/pull/2778

ALTree, 2022-04-15:
> does this test case come out of a fuzzer?

Yes.

closed 2022-04-29 by deadprogram