tinygo-org/tinygo #2779: tinygo:ld.lld: error: undefined symbol: crypto/aes.encryptBlockAsm

d33tah, 2022-04-15:
I tried to give tinygo a try by building a simple "hello world" web service. Here's what I got in Docker:

```
> sudo docker run --rm --entrypoint=/bin/bash -ti tinygo/tinygo:0.22.0
root@13d715ef4d83:/# cat > main.go
package main

import (
    "fmt"
    "net/http"
    "log"
)

func main() {

    // handle `/` route to `http.DefaultServeMux`
    http.HandleFunc( "/", func( res http.ResponseWriter, req *http.Request ) {

        // get response headers
        header := res.Header()

        // set content type header
        header.Set("Content-Type","application/json")

        // reset date header (inline call)
        res.Header().Set("Date","01/01/2020")
        
        // set status header
        res.WriteHeader(http.StatusBadRequest) // http.StatusBadRequest == 400
        
        // respond with a JSON string
        fmt.Fprint(res, `{"status":"FAILURE"}`)
    } )

    // listen and serve using `http.DefaultServeMux`
    log.Fatal(http.ListenAndServe(":9000", nil))

}
root@13d715ef4d83:/# tinygo build -o main main.go
tinygo:ld.lld: error: undefined symbol: crypto/aes.encryptBlockAsm
>>> referenced by aes_gcm.go:119 (/usr/local/go/src/crypto/aes/aes_gcm.go:119)
>>>               /tmp/tinygo3252666005/main.o:(interface:{NonceSize:func:{}{basic:int},Open:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8,named:error},Overhead:func:{}{basic:int},Seal:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8}}.Seal$invoke)
>>> referenced by aes_gcm.go:172 (/usr/local/go/src/crypto/aes/aes_gcm.go:172)
>>>               /tmp/tinygo3252666005/main.o:(interface:{NonceSize:func:{}{basic:int},Open:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8,named:error},Overhead:func:{}{basic:int},Seal:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8}}.Open$invoke)
>>> referenced by cipher_asm.go:69 (/usr/local/go/src/crypto/aes/cipher_asm.go:69)
>>>               /tmp/tinygo3252666005/main.o:((*crypto/aes.aesCipherAsm).Encrypt)

tinygo:ld.lld: error: undefined symbol: crypto/aes.gcmAesData
>>> referenced by aes_gcm.go:122 (/usr/local/go/src/crypto/aes/aes_gcm.go:122)
>>>               /tmp/tinygo3252666005/main.o:(interface:{NonceSize:func:{}{basic:int},Open:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8,named:error},Overhead:func:{}{basic:int},Seal:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8}}.Seal$invoke)
>>> referenced by aes_gcm.go:175 (/usr/local/go/src/crypto/aes/aes_gcm.go:175)
>>>               /tmp/tinygo3252666005/main.o:(interface:{NonceSize:func:{}{basic:int},Open:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8,named:error},Overhead:func:{}{basic:int},Seal:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8}}.Open$invoke)

tinygo:ld.lld: error: undefined symbol: crypto/aes.gcmAesEnc
>>> referenced by aes_gcm.go:129 (/usr/local/go/src/crypto/aes/aes_gcm.go:129)
>>>               /tmp/tinygo3252666005/main.o:(interface:{NonceSize:func:{}{basic:int},Open:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8,named:error},Overhead:func:{}{basic:int},Seal:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8}}.Seal$invoke)

tinygo:ld.lld: error: undefined symbol: vendor/golang.org/x/crypto/chacha20poly1305.chacha20Poly1305Seal
>>> referenced by chacha20poly1305_amd64.go:62 (/usr/local/go/src/vendor/golang.org/x/crypto/chacha20poly1305/chacha20poly1305_amd64.go:62)
>>>               /tmp/tinygo3252666005/main.o:(interface:{NonceSize:func:{}{basic:int},Open:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8,named:error},Overhead:func:{}{basic:int},Seal:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8}}.Seal$invoke)

tinygo:ld.lld: error: undefined symbol: crypto/aes.gcmAesFinish
>>> referenced by aes_gcm.go:131 (/usr/local/go/src/crypto/aes/aes_gcm.go:131)
>>>               /tmp/tinygo3252666005/main.o:(interface:{NonceSize:func:{}{basic:int},Open:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8,named:error},Overhead:func:{}{basic:int},Seal:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8}}.Seal$invoke)
>>> referenced by aes_gcm.go:184 (/usr/local/go/src/crypto/aes/aes_gcm.go:184)
>>>               /tmp/tinygo3252666005/main.o:(interface:{NonceSize:func:{}{basic:int},Open:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8,named:error},Overhead:func:{}{basic:int},Seal:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8}}.Open$invoke)

tinygo:ld.lld: error: undefined symbol: crypto/aes.gcmAesDec
>>> referenced by aes_gcm.go:182 (/usr/local/go/src/crypto/aes/aes_gcm.go:182)
>>>               /tmp/tinygo3252666005/main.o:(interface:{NonceSize:func:{}{basic:int},Open:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8,named:error},Overhead:func:{}{basic:int},Seal:func:{slice:basic:uint8,slice:basic:uint8,slice:basic:uint8,slice:basic:uint8}{slice:basic:uint8}}.Open$invoke)
[... 83 lines cut ...]
>>> referenced by p256_asm.go:356 (/usr/local/go/src/crypto/elliptic/p256_asm.go:356)
>>>               /tmp/tinygo3252666005/main.o:((*crypto/elliptic.p256Point).p256PointToAffine)
>>> referenced 11 more times

tinygo:ld.lld: error: undefined symbol: crypto/elliptic.p256FromMont
>>> referenced by p256_asm.go:319 (/usr/local/go/src/crypto/elliptic/p256_asm.go:319)
>>>               /tmp/tinygo3252666005/main.o:((*crypto/elliptic.p256Point).p256PointToAffine)
>>> referenced by p256_asm.go:320 (/usr/local/go/src/crypto/elliptic/p256_asm.go:320)
>>>               /tmp/tinygo3252666005/main.o:((*crypto/elliptic.p256Point).p256PointToAffine)

tinygo:ld.lld: error: undefined symbol: crypto/elliptic.p256LittleToBig
>>> referenced by p256_asm.go:324 (/usr/local/go/src/crypto/elliptic/p256_asm.go:324)
>>>               /tmp/tinygo3252666005/main.o:((*crypto/elliptic.p256Point).p256PointToAffine)
>>> referenced by p256_asm.go:325 (/usr/local/go/src/crypto/elliptic/p256_asm.go:325)
>>>               /tmp/tinygo3252666005/main.o:((*crypto/elliptic.p256Point).p256PointToAffine)
>>> referenced by p256_asm.go:192 (/usr/local/go/src/crypto/elliptic/p256_asm.go:192)
>>>               /tmp/tinygo3252666005/main.o:((crypto/elliptic.p256Curve).Inverse)

tinygo:ld.lld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
failed to run tool: ld.lld
error: failed to link /tmp/tinygo3252666005/main: exit status 1
root@13d715ef4d83:/# 
```

It looks like those packages are on the list of the ones [supported by TinyGo](https://tinygo.org/docs/reference/lang-support/stdlib/), so what's the problem?

dkegel-fastly, 2022-04-15:
You need to read the large type at the top of that page :-)

A PR is pending to update that page to show which packages actually pass tests, which ought to help clear up the confusion.

The tests that pass in the dev branch (which will soon become the next version, 0.23) on darwin, linux, wasi, and windows can be found by checking out that branch and typing "make report-stdlib-tests-pass".  Here's what it currently says:
```
[report.txt](https://github.com/tinygo-org/tinygo/files/8496646/report.txt)

d33tah, 2022-04-15:
Got it. Thanks for pointing that out! I guess that the issue can be closed then.

dkegel-fastly, 2022-04-15:
It's a real enough bug.  Easier way to demonstrate it is:

```
C02DG0QJMD6R:tinygo dkegel$ build/tinygo test crypto/aes
tinygo:ld.lld: error: undefined symbol: _crypto/aes.encryptBlockAsm
>>> referenced by /var/folders/v0/0k9hwftd29b18z9z7s2g2sfc0000gn/T/tinygo2654768174/main.o

tinygo:ld.lld: error: undefined symbol: _crypto/aes.decryptBlockAsm
>>> referenced by /var/folders/v0/0k9hwftd29b18z9z7s2g2sfc0000gn/T/tinygo2654768174/main.o
failed to run tool: ld.lld
error: failed to link /var/folders/v0/0k9hwftd29b18z9z7s2g2sfc0000gn/T/tinygo2654768174/main: exit status 1
FAIL
```

You might tidy up this bug report by editing down the long log in your initial comment, though.

dkegel-fastly, 2022-04-15:
btw, the missing symbol comes from regular go, and isn't picked up because it is in assembly and we don't quite support that yet; https://github.com/tinygo-org/tinygo/pull/2688 may eventually fix that.

d33tah, 2022-04-15:
> You might tidy up this bug report by editing down the long log in your initial comment, though.

@dkegel-fastly done. does it look OK now?

aykevl, 2022-08-28:
This issue would be fixed with #3103.

dkegel-fastly, 2022-09-20:
Also affects go-jose.

dgryski, 2024-10-21:
Upstream Go now has `purego` tags for AES (and other crypto packages) which we are able to take advantage of

```
~/go/src/github.com/tinygo-org/tinygo $ tinygo test crypto/aes
ok  	crypto/aes	0.533s
```

closed 2024-10-21 by dgryski