tinygo-org/tinygo #2795: tinygo/lib/compiler-rt/lib/builtins/absvdi2.c: could not locate Clang headers

arbourd, 2022-04-23:
Strange new behaviour when building an arm64 container from source (which worked fine prior to `v0.22.0`). I don't have any experience with `clang` or `llvm` so I'd love some help understanding why this is happening.

Failing on `make release` after running a successful `make`.

aykevl, 2022-04-23:
Can you tell a bit more about the environment? OS/arch/etc

arbourd, 2022-04-23:
Absolutely! Thank for taking the time @aykevl.

I got it running on both `amd64` and `arm64` using a different Dockerfile (for our purposes). It is here if that helps. It is located here: https://github.com/suborbital/subo/blob/6e49f277e7c064c9f8a2de08b878adb0c66c9c9b/builder/docker/tinygo/Dockerfile.base#L28-L29

We're purposely skipping some other steps and with `v0.21.0` we've had no issues building. I'm guessing it might have been [this change](https://github.com/tinygo-org/tinygo/commit/3883550c446fc8b14115366284316290f3d3c4b1) that is causing issues, I just don't understand why the clang headers would have moved 😅

aykevl, 2022-04-27:
Actually, this might be pretty simple to solve. Can you patch the lines here to be `./build/release/tinygo/bin/tinygo` instead of `./build/tinygo` and see whether that fixes the issue?

https://github.com/tinygo-org/tinygo/blob/3883550c446fc8b14115366284316290f3d3c4b1/Makefile#L644-L649

arbourd, 2022-04-27:
that seems to have worked @aykevl! thank you.

should we patch it this way in `dev` as well? whats the justification for using the "regular build location"?

arbourd, 2022-05-04:
Ah I understand what's going on now, I just needed to move the headers from `clang/lib/Headers`

aykevl, 2022-05-05:
> should we patch it this way in `dev` as well? whats the justification for using the "regular build location"?

If you want to make a PR, that would be great!

commit af6bbaf5 (2022-05-19): Use release built tinygo in Makefile

Closes #2795