tinygo-org/tinygo #2798: message: '-pacbti' is not a recognized feature for this target (ignoring feature)

oflebbe, 2022-04-23:
When compiling for an rp2040 (with llvm13) on current dev
I get the message 

` '-pacbti' is not a recognized feature for this target (ignoring feature)`

I guess this line is the one triggering it.
https://github.com/tinygo-org/tinygo/blob/ee497a296b8afcde3f8b15fb68223818e5a2dda6/targets/cortex-m0plus.json#L5

However if I check https://community.arm.com/arm-community-blogs/b/architectures-and-processors-blog/posts/armv8-1-m-pointer-authentication-and-branch-target-identification-extension this feature is an ARM8 extension and isn't available on 32bit targets. Is this added accidentely ? @aykevl ?

aykevl, 2022-04-25:
It has a `-` in front of it so it disables this feature. So that's correct.

This feature is new in LLVM 14. Updating to LLVM 14 (build tinygo with `-tags=llvm14`) will get the feature recognized and will let the warning disappear. I've added `-pacbti` to match Clang because if it doesn't match Clang, LLVM refuses to inline such functions between C and Go.

ysoldak, 2022-04-25:
I think we have to wait for `llvm@14` on Mac to test that.
There is an open PR https://github.com/Homebrew/homebrew-core/pull/97618

QuLogic, 2022-05-08:
> This feature is new in LLVM 14.

I suppose this means I can patch out the `-pacbti` on our LLVM 13 builds then?

aykevl, 2022-05-08:
Yes, that's fine.

(I know it's not a great solution, but I'm not sure what's the best way to fix this).

QuLogic, 2022-08-07:
For LLVM 13, what I did was https://github.com/QuLogic/tinygo/commit/1e77b52409a8262b81c11aba18401ba61b4fd2bf, which you may want to use for packages.

However, #3039 has dropped support for LLVM 13, so I guess this can be closed.

closed 2022-08-07 by QuLogic

aykevl, 2022-08-07:
> For LLVM 13, what I did was [QuLogic@1e77b52](https://github.com/QuLogic/tinygo/commit/1e77b52409a8262b81c11aba18401ba61b4fd2bf), which you may want to use for packages.

That's a workaround if you encounter a similar issue again, but note that it if this patch is applied for LLVM 14 as well it will block inlining between C and Go code. That might not be what you intended.

QuLogic, 2022-08-07:
Yes, it's only for LLVM 13.