tinygo-org/tinygo #2800: "tinygo test -target wasi compress/lzw" fails on windows?

dankegel, 2022-04-23:
Using today's dev branch plus a fix for #2799, on a (relatively) freshly set up windows 11 system, compress/lzw seems to fail on wasi:
```
$ ~/src/tinygo/build/tinygo test -target wasi -v compress/lzw
FAIL    compress/lzw    16.797s
```

deadprogram, 2022-06-12:
Is this still an issue?

dulanov, 2022-07-19:
By me:

```shell
C:\>tinygo test -target wasi -v compress/lzw                                                                                                                                                                                                                         error: wasm-opt failed: exit status 0xc0000409                                                                                                                                                                                                                                 FAIL                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                          

C:>tinygo version                                                                                                                                                                                                                                                   tinygo version 0.24.0 windows/amd64 (using go version go1.18.4 and LLVM version 14.0.0)
```

dgryski, 2022-07-19:
@dulanov That seems like a different error than the one @dankegel reported.

dgryski, 2023-02-02:
`compress/lzw` isn't playing nicely with wasi at all.

https://github.com/tinygo-org/tinygo/issues/2525

Closing as a dupe for now.

closed 2023-02-02 by dgryski