tinygo-org/tinygo #2806: (constant -4 of type uintptr) overflows uintptr

kyleconroy, 2022-04-25:
TinyGo can't compiled the the [pointer_unsafe.go](https://github.com/protocolbuffers/protobuf-go/blob/master/internal/impl/pointer_unsafe.go#L142) file from https://github.com/protocolbuffers/protobuf-go. I get the following error message:

```
unsafe.Sizeof(unsafe.Pointer(nil)) - unsafe.Sizeof((MessageState literal)) (constant -4 of type uintptr) overflows uintptr
```

aykevl, 2022-04-25:
This comment above it is relevant:

```
// Static check that MessageState does not exceed the size of a pointer.
```

So it appears that `MessageState` is larger than a pointer.
Without further information, this would be hard to debug. How can I reproduce this?

---

That said, this is most certainly unsafe code so that throws away all promises of portability. But quickly looking at it, I don't see why it fails.

skabbes, 2022-04-25:
My guess is that possibly TinyGo (or maybe the specific architecutre you are targeting) is considering a `uintptr` to be unsigned (the `u` in uintptr), and therefore -4 is an invalid value (or it is wrapping around to a very very large number).

skabbes, 2022-04-25:
I had a comment previously that I deleted, I did not read aykevl's comment closely enough.

The [MessageState](https://github.com/protocolbuffers/protobuf-go/blob/8a7ba0762cb3b39fc0536379eac2f7fa5796f187/internal/impl/message_reflect.go#L360) struct embeds a [few structs](https://github.com/protocolbuffers/protobuf-go/blob/8a7ba0762cb3b39fc0536379eac2f7fa5796f187/internal/pragma/pragma.go#L11) with 0 fields.

My _new_ guess is that these structs take space in TinyGo, but do not take space in "regular Go".

kyleconroy, 2022-04-26:
> Without further information, this would be hard to debug. How can I reproduce this?

Sorry about that. Here's the reproduction.

```sh
go mod init github.com/tinygo-issues
go get google.golang.org/protobuf
```

```go
package main

import (
	"fmt"

	_ "google.golang.org/protobuf/runtime/protoimpl"
)

func main() {
	fmt.Println("protoc")
}
```

```sh
tinygo build -o example.wasm -gc=leaking -scheduler=none -wasm-abi=generic -target=wasi
# google.golang.org/protobuf/internal/impl
../../go/pkg/mod/google.golang.org/protobuf@v1.28.0/internal/impl/pointer_unsafe.go:142:16: unsafe.Sizeof(unsafe.Pointer(nil)) - unsafe.Sizeof((MessageState literal)) (constant -4 of type uintptr) overflows uintptr
make: *** [run] Error 1
```

skabbes, 2022-05-18:
I trimmed down the minimal repro - to be something even more minimal.

I think the bug can be restated as "sizeof 0-length arrays of functions is greater than 0.

```go
package main

import (
	"unsafe"
)

// DoNotCompare can be embedded in a struct to prevent comparability.
// This struct, when embedded takes non-zero space
type DoNotCompare [0]func()

type MessageState struct {
	DoNotCompare
	data *int32
}

// Static check that MessageState does not exceed the size of a pointer.
const _ = unsafe.Sizeof(unsafe.Pointer(nil)) - unsafe.Sizeof((MessageState{}))

func main() {
	println("protoc minimal example")
}

```

skabbes, 2022-05-19:
FYI @kyleconroy https://github.com/tinygo-org/tinygo/pull/2854 fixed this issue, it is merged into the dev branch.  I think you can probably close this issue.

closed 2022-07-01 by deadprogram