tinygo-org/tinygo #2808: tinygo test -v hoards output?

dkegel-fastly, 2022-04-25:
https://dave.cheney.net/2020/03/10/go-test-v-streaming-output points out that go 1.14 arranged for go test -v to stream output immediately rather than hoarding it.

I suspect I broke this property in tinygo when I added support for subtests.

dgryski, 2022-04-25:
I had a hacky patched version of this in my local tree for a while to let me see which individual test was timing out because the entire output was being buffered, and then discarded when the test binary crashed.  It's probably not too hard to put together nicely.

deadprogram, 2023-05-23:
I think this was addressed by https://github.com/tinygo-org/tinygo/pull/3625 so tagging to close on next release.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing this issue. Thanks!

closed 2023-06-14 by deadprogram