tinygo-org/tinygo #2815: unsuccessful in installing  tinygo on Raspberry pi 4 Rasbian OS 64bit

hemuvemula, 2022-04-30:
### How did i install tinygo..?
followed https://tinygo.org/getting-started/install/linux/#ubuntu-debian
Step1: 
`wget https://github.com/tinygo-org/tinygo/releases/download/v0.23.0/tinygo_0.23.0_armhf.deb`
Step 2:
`sudo dpkg -i tinygo_0.23.0_armhf.deb`
Step 3:
`export PATH=$PATH:/usr/local/bin
`
### how did i determine installation failed:
tinygo version command failed 
`pi@raspberrypi:~ $ tinygo version
-bash: /usr/local/bin/tinygo: No such file or directory`

verified that tinygo is in /use/local/bin and also that the path is set appropriately 

`pi@raspberrypi:~ $ which tinygo
/usr/local/bin/tinygo`

`pi@raspberrypi:/usr/local/bin $ ls -lrt
total 0
lrwxrwxrwx 1 root root 24 Apr 28 23:06 tinygo -> ../lib/tinygo/bin/tinygo`

***********************************************************

### os info:
`pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"`

### rasberrypi proc info 

`pi@raspberrypi:~ $ cat /proc/version
Linux version 5.15.32-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022`

`pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03115
Serial		: 100000003d7f3362
Model		: Raspberry Pi 4 Model B Rev 1.5`

dankegel, 2022-05-01:
FWIW, armhf is for 32 bit support.  Since you're on a 64 bit OS, you probably want  https://github.com/tinygo-org/tinygo/releases/download/v0.23.0/tinygo_0.23.0_amd64.deb instead.

On my old raspberry pi 4 with a 32 bit os, the hf package seems to run, at least, 'tinygo version' prints out the right thing.

aykevl, 2022-05-01:
> FWIW, armhf is for 32 bit support. Since you're on a 64 bit OS, you probably want https://github.com/tinygo-org/tinygo/releases/download/v0.23.0/tinygo_0.23.0_amd64.deb instead.

That's amd64, not arm64. That's a very big difference.

What's really going on is that you're running an armhf (32-bit ARM) binary on a 64-bit ARM system (arm64, aarch64). The "file not found" does not refer to the `tinygo` binary itself but rather to dynamically linked libraries. You can find out which ones you're missing with the following command:

    ldd /usr/local/bin/tinygo

dankegel, 2022-05-01:
It took me a while to get the 64 bit os up (rpi-imager kept failing with an fsync error until I creatde a linux partition on the sdk card).

Try
  sudo apt install libstdc++6:armhf
That seems to be enough to let the 32 bit tinygo run on the current 64 bit raspberry pi debian derivative.

deadprogram, 2022-06-12:
The next release will have 64-bit executable.

PR #2826 (2022-07-08): ci: build Linux binary in Alpine container
This makes it easier to move the TinyGo compiler between Linux versions because it doesn't depend on any system libraries anymore. For example, binaries should be able to run on old Linux versions and on distributions without glibc (such as Alpine Linux).

This doesn't directly fix #2815 but a similar fix for ARM would have avoided the #2815 issue.
closed 2022-07-01 by deadprogram