tinygo-org/tinygo #2816: "tinygo test log" fails because runtime.Caller(2) does not return path to source file

dankegel, 2022-05-01:
Symptom:

```
--- FAIL: TestAll (0.00s)
    log output should match "^.*/[A-Za-z0-9_\\-]+\\.go:(60|62): hello 23 world$" is "???:0: hello 23 world"
    ...
```

Just to illustrate what "tinygo test log" is looking for, the following ugly kludge makes it pass:
```
 func Caller(skip int) (pc uintptr, file string, line int, ok bool) {
+        // Hack demonstrating what log's test wants
+        if skip == 2 {
+                return 0, "/nul/unknown.go", 60, true
+        }
        return 0, "", 0, false
 }
```

dkegel-fastly, 2022-08-22:
It is tempting to hope that at least this could be supported:

    _, filename, _, ok := runtime.Caller(0)

My usecase for that is "sprinkling debug prints throughout the source" :-)

aykevl, 2022-09-08:
That would certainly be interesting to support! I think it is possible, using libunwind and libbacktrace (as long as you don't strip debug information from the binary, and it might not work at all on wasm).