tinygo-org/tinygo #2817: "tinygo test io/ioutil" fails because os.PathError has not moved to io/fs as in upstream

dankegel, 2022-05-01:
Currently, "tinygo test io/ioutil" fails like this post go 1.15:
```
--- FAIL: TestTempDir_BadDir (0.00s)
    TempDir error = &T{Op:"stat", Path:"/tmp/TestTempDir_BadDir216851801/not-exist", Err:(T)(0x247638)}; want PathError for path "/tmp/TestTempDir_BadDir216851801/not-exist" satisifying os.IsNotExist
```
while the nearly identical test in os,  TestMkdirTempBadDir, passes.

See this comment in os/file.go:
```
// TODO: PathError moved to io/fs in go 1.16 and left an alias in os/errors.go.
// Do the same once we drop support for go 1.15.
type PathError struct {
```

Patch coming.

deadprogram, 2022-05-18:
I think this can be marked to be closed on next release, is that correct @dankegel ?

dkegel-fastly, 2022-05-26:
Quite.

PR #2825 (2022-05-03): Drop support for go 1.15
1.15 specific files deleted.
1.16 specific files folded carefully into generic files, with goal of reducing diff with upstream.
Follows upstream 1.16 in making PathError etc. be aliases for the same errors in io/fs.

This fixes #2817 and lets us add io/ioutil to "make test-tinygo" on linux and mac.
commit e87cd23e (2022-05-03): os: Drop support for go 1.15

1.15 specific files deleted.
1.16 specific files folded carefully into generic files, with goal of reducing diff with upstream.
Follows upstream 1.16 in making PathError etc. be aliases for the same errors in io/fs.

This fixes #2817 and lets us add io/ioutil to "make test-tinygo" on linux and mac.