tinygo-org/tinygo #2822: armhf 0.23 build doesn't recognize keyword 'any'?

dankegel, 2022-05-01:
On a raspberry pi 4 running an older 32 bit version of raspbian, installing the latest go and tinygo from their respective upstream download directories yields an installation that kind of works, but does not recognize the 'any' keyword:

```
pi@rbb-rpi4-1:~/local $ uname -a
Linux rbb-rpi4-1 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux
pi@rbb-rpi4-1:~/local $ go version
go version go1.18.1 linux/arm
pi@rbb-rpi4-1:~/local $ tinygo version
tinygo version 0.23.0 linux/arm (using go version go1.18.1 and LLVM version 14.0.0)
pi@rbb-rpi4-1:~/local $ tinygo test os
# errors
/home/pi/local/go/src/errors/wrap.go:78:27: undeclared name: any
/home/pi/local/go/src/errors/wrap.go:96:34: undeclared name: any
FAIL
```

aykevl, 2022-05-01:
See https://github.com/tinygo-org/tinygo/pull/2823

I think we can just add the build artifact from there to the current release because it branches off from the release branch and so it's identical to the release except for the Go version.

deadprogram, 2022-05-02:
I have updated the 0.23.0 release with the build artifacts from #2823 

Now closing this since the issue has been resolved in the current release.

Please re-open if needed, thank you everyone!

closed 2022-05-02 by deadprogram