tinygo-org/tinygo #2824: armhf linux build fails "tinygo test os" in TestDirFS?

dankegel, 2022-05-01:
On raspberry pi 4 64 bits, I installed https://github.com/tinygo-org/tinygo/releases/download/v0.23.0/tinygo_0.23.0_armhf.deb (and the right libc++, see #2815 ) and go 1.17, and ran "tinygo test os".

The test fails with 
```
--- FAIL: TestDirFS (0.00s)
    TestFS found errors:
        b: failed TestReader:
        	Seek(0, 1) from EOF = 0, operation not supported, want 0, nil
        dir/x: failed TestReader:
        	Seek(0, 1) from EOF = 0, operation not supported, want 0, nil
        a: failed TestReader:
        	Seek(0, 1) from EOF = 0, operation not supported, want 0, nil
    FailNow is incomplete, requires runtime.Goexit()
```

Same result with go 1.18 (with the fixed tinygo binary from #2823).

QuLogic, 2022-05-08:
This also [fails on i686](https://koji.fedoraproject.org/koji/taskinfo?taskID=86789232), so seems like maybe a 32-bit problem in general?

QuLogic, 2022-05-08:
In `os_test.go`, I see
```
func TestSeek(t *testing.T) {
        if runtime.GOARCH == "386" || runtime.GOARCH == "arm" {
                t.Log("TODO: implement seek for 386 and arm")
                return
        }
```
so this probably needs to be translated into `TestFS` as well, but I don't think there's a copy of it in tinygo that can be patched.

QuLogic, 2022-05-08:
I also see
```
--- FAIL: TestSymlink (0.00s)
    TestFS found errors:
        hello: failed TestReader:
        	Seek(0, 1) from EOF = 0, operation not supported, want 13, nil
        hello.link: failed TestReader:
        	Seek(0, 1) from EOF = 0, operation not supported, want 13, nil
    FailNow is incomplete, requires runtime.Goexit()
FAIL
```
but this one I cannot figure how to get from `TestSymlink` to a `Seek`.

QuLogic, 2022-05-08:
Ah, the latter is from `testing/fstest`; I worked around that by skipping it on i686.

deadprogram, 2022-05-31:
Now that we have 64-bit ARM Linux build, it this still a problem?

dankegel, 2022-06-01:
Well... Yeah, it's still a problem for 32 bit systems, and will be until tinygo supports go's assembly implementation of seek,
https://github.com/golang/go/blob/master/src/syscall/asm_linux_arm.s

dankegel, 2022-06-01:
Skipping particular tests on affected systems using build tags might be a fine bandaid.  I dunno why I didn't do that when I was fleshing out tinygo's is package, guess I was lazy

aykevl, 2022-09-01:
This will be fixed by #3103.

closed 2026-11-05 by deadprogram