tinygo-org/tinygo #2832: panic in golang.org/x/tools/go/types/typeutil.Hasher.hashFor when compiling generics code

yoursunny, 2022-05-06:
**main.go**

```go
package main

func F[T any](x T) {
}

func main() {
	F(1)
}
```

Commands and error message:

```bash
yoursunny@localhost:~/snippet$ go version
go version go1.18.1 linux/amd64

yoursunny@localhost:~/snippet$ tinygo version
tinygo version 0.23.0 linux/amd64 (using go version go1.18.1 and LLVM version 14.0.0)

yoursunny@localhost:~/snippet$ tinygo build -target=wasm main.go
panic: T

goroutine 29 [running]:
golang.org/x/tools/go/types/typeutil.Hasher.hashFor({0x8?}, {0x54236c0?, 0xc0001f6a20?})
        /home/runner/go/pkg/mod/golang.org/x/tools@v0.1.6-0.20210813165731-45389f592fe9/go/types/typeutil/map.go:302 +0x41f
golang.org/x/tools/go/types/typeutil.Hasher.Hash({0x203000?}, {0x54236c0, 0xc0001f6a20})
        /home/runner/go/pkg/mod/golang.org/x/tools@v0.1.6-0.20210813165731-45389f592fe9/go/types/typeutil/map.go:226 +0x65
golang.org/x/tools/go/types/typeutil.Hasher.hashTuple({0x0?}, 0xc00093b6b0)
        /home/runner/go/pkg/mod/golang.org/x/tools@v0.1.6-0.20210813165731-45389f592fe9/go/types/typeutil/map.go:310 +0x5f
golang.org/x/tools/go/types/typeutil.Hasher.hashFor({0x30?}, {0x5423620?, 0xc0007f5800?})
        /home/runner/go/pkg/mod/golang.org/x/tools@v0.1.6-0.20210813165731-45389f592fe9/go/types/typeutil/map.go:276 +0x10c
golang.org/x/tools/go/types/typeutil.Hasher.Hash({0x0?}, {0x5423620, 0xc0007f5800})
        /home/runner/go/pkg/mod/golang.org/x/tools@v0.1.6-0.20210813165731-45389f592fe9/go/types/typeutil/map.go:226 +0x65
golang.org/x/tools/go/types/typeutil.(*Map).Set(0xc0000bb138, {0x5423620?, 0xc0007f5800}, {0x53349a0?, 0x74b0660})
        /home/runner/go/pkg/mod/golang.org/x/tools@v0.1.6-0.20210813165731-45389f592fe9/go/types/typeutil/map.go:123 +0x114
golang.org/x/tools/go/ssa.(*Program).needMethods(0xc0000bb0e0, {0x5423620?, 0xc0007f5800?}, 0x0)
        /home/runner/go/pkg/mod/golang.org/x/tools@v0.1.6-0.20210813165731-45389f592fe9/go/ssa/methods.go:162 +0xcf
golang.org/x/tools/go/ssa.(*Program).needMethodsOf(0xc0000bb0e0, {0x5423620?, 0xc0007f5800?})
        /home/runner/go/pkg/mod/golang.org/x/tools@v0.1.6-0.20210813165731-45389f592fe9/go/ssa/methods.go:145 +0x70
golang.org/x/tools/go/ssa.(*Package).build(0xc0002842a0)
        /home/runner/go/pkg/mod/golang.org/x/tools@v0.1.6-0.20210813165731-45389f592fe9/go/ssa/builder.go:2281 +0x111
sync.(*Once).doSlow(0x5346ac0?, 0x16?)
        /opt/hostedtoolcache/go/1.18.1/x64/src/sync/once.go:68 +0xc2
sync.(*Once).Do(...)
        /opt/hostedtoolcache/go/1.18.1/x64/src/sync/once.go:59
golang.org/x/tools/go/ssa.(*Package).Build(...)
        /home/runner/go/pkg/mod/golang.org/x/tools@v0.1.6-0.20210813165731-45389f592fe9/go/ssa/builder.go:2269
github.com/tinygo-org/tinygo/compiler.CompilePackage({0xc0001b60f0?, 0x59?}, 0xc00019ae00, 0xc0002842a0, {0x5?}, 0x0?, 0x0?)
        /home/runner/work/tinygo/tinygo/compiler/compiler.go:258 +0x1c5
github.com/tinygo-org/tinygo/builder.Build.func1(0x0)
        /home/runner/work/tinygo/tinygo/builder/build.go:289 +0x191
github.com/tinygo-org/tinygo/builder.runJob(0xc0002845a0, 0x0?)
        /home/runner/work/tinygo/tinygo/builder/jobs.go:222 +0x4f
created by github.com/tinygo-org/tinygo/builder.runJobs
        /home/runner/work/tinygo/tinygo/builder/jobs.go:123 +0x5db
```

deadprogram, 2022-05-06:
Generics are not yet supported in TinyGo.

deadprogram, 2022-06-11:
See https://github.com/tinygo-org/tinygo/pull/2903

closed 2022-07-01 by deadprogram