tinygo-org/tinygo #2845: XIAO has no UART0

tradeJmark, 2022-05-10:
From the documentation for the Seeeduino XIAO:

> You can use the USB port to the XIAO as a serial port. `UART0` refers to this connection.

But if you try to use `UART0` with the `xiao` target, say using the [echo example](https://github.com/tinygo-org/tinygo/blob/master/src/examples/echo/echo.go), you get `UART0 not declared by package machine`. You acutally can use `println` and it will output over the USB serial port, but I need to read input. Is the doc wrong, or do I need a modified echo program or something?

sago35, 2022-05-10:
The documentation seems to be wrong.
The correct definition is UART1.

https://tinygo.org/docs/reference/microcontrollers/xiao/

sago35, 2022-05-10:
It may be a good idea to update examples/echo as well, since different boards have different definitions.
Maybe we should try to use machine.DefaultUART.
However, in the case of XIAO, machine.DefaultUART is not set, so I think it needs to be addressed.

tradeJmark, 2022-05-10:
`UART1` looks like it references the UART interface on pins D6 and D7, which I don't believe is wired to the USB interface. The correct pins for the USB serial interface actually are referenced in /src/machine/board_xiao.go, they're named `USBCDC_DM_PIN` and `USBCDC_DP_PIN`, but they don't seem to be linked to an actual interface object as far as I can tell.

sago35, 2022-05-10:
Perhaps you can use machine.Serial.

tradeJmark, 2022-05-11:
That seems to work.

deadprogram, 2022-05-18:
Is there an action item on this issue, or should it now be closed?

closed 2022-08-04 by deadprogram