tinygo-org/tinygo #2846: Can you use tinygo just for the LLVM size optimizations as part of normal go builds?

lanza, 2022-05-10:
As the title states, I'm interested in using tinygo's LLVM based compiler just for the size optimizations that go lacks for normal go builds. Currently I'm running into an issue with `TCPAddr` missing from the `net` package. AFAIK tinygo is using the stdlib library from `/usr/lib64/tinygo/src/net`. Is there anyway to just enforce the normal stdlib with the tinygo compiler?

dgryski, 2022-05-10:
No, I don't believe so. The compiler and runtime are tightly integrated, and that's true for both the tinygo project and the main Go project.

closed 2022-05-11 by dkegel-fastly

aykevl, 2022-05-12:
@lanza TinyGo can't do this. But you may be interested in gollvm:
https://go.googlesource.com/gollvm/

aykevl, 2022-05-12:
Also, TinyGo also supports Linux/macOS/Windows as a compilation target, it's not just limited to embedded systems.

lanza, 2022-05-13:
> @lanza TinyGo can't do this. But you may be interested in gollvm:
https://go.googlesource.com/gollvm/

Interesting. This was removed from llvm due to it being dead for years, but evidently people have been supporting it on that googlesource repo. Thanks for the link.

> Also, TinyGo also supports Linux/macOS/Windows as a compilation target, it's not just limited to embedded systems.

Yup, this was my motivation in asking. I figured if it could target everything go could then it probably doesn't have a hard coupling to the runtime. I was hoping there was some `--use-go-runtime` flag override. 

> No, I don't believe so. The compiler and runtime are tightly integrated, and that's true for both the tinygo project and the main Go project.

How so? The runtime is clearly part of the project, but looking over the commits that added the runtime files from go, they seem like a copy/paste job. Do you know of a sticking point that would cause an issue?