tinygo-org/tinygo #2849: error: failed to build /usr/lib64/tinygo/src/device/arm/cortexm.s: %#v: none of these commands were found in your $PATH: clang-13

s-fairchild, 2022-05-15:
I am attempting to use the BBC Microbit and it's failing to compile:
```bash
tinygo flash -target=microbit-v2 -programmer=cmsis-dap .
error: failed to build /usr/lib64/tinygo/src/device/arm/cortexm.s: %#v: none of these commands were found in your $PATH: clang-13
```

System is Fedora 36 from the official Fedora repos
```bash
[steven@pc]$ rpm -q golang tinygo
golang-1.18-1.fc36.x86_64
tinygo-0.22.0-7.fc36.x86_64
```

Here's the exact board I'm using: https://www.amazon.com/gp/product/B01G8WUGWU/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Here are some other targets I tried as well, all with the same error but different build path.
```bash
tinygo flash -target=microbit-v2-s113v7 -programmer=cmsis-dap .
tinygo flash -target=microbit-s110v8 -programmer=cmsis-dap .
tinygo flash -target=microbit-v2 -programmer=cmsis-dap .
tinygo flash -target=microbit -programmer=cmsis-dap .
tinygo flash -target=microbit-v2-s113v7 -programmer=cmsis-dap .
tinygo flash -target=microbit-v2 -programmer=cmsis-dap .
tinygo flash -target=microbit-v2 .
tinygo flash -target=microbit-v2-s113v7 .
tinygo flash -target=microbit .
tinygo flash -target=microbit-s110v8 .
tinygo flash -target=microbit-v2 -programmer=cmsis-dap .
tinygo flash -target=microbit-v2 -programmer=cmsis-dap .
```

s-fairchild, 2022-05-15:
clang-13 doesn't exist in Fedora 36 repos, only clang-14.
Resolved by creating a symbolic link from `clang` to `clang-13`
```bash
ln -s /usr/bin/clang /usr/bin/clang-13
```

closed 2022-05-15 by s-fairchild

aykevl, 2022-05-15:
@QuLogic this looks like a Fedora packaging bug

QuLogic, 2022-05-22:
Darn, people upgraded too quick!

I was having issues with 0.23.0 on i686 crashing, but LLVM or Clang or something was updated and fixed it. The update is https://bodhi.fedoraproject.org/updates/FEDORA-2022-0f4fad9670