tinygo-org/tinygo #2851: WebAssembly: indicate initialization status of runtime

jzabinski-dolios, 2022-05-16:
For WebAssembly, the embedder needs to use `wasm_exec`'s `run(instance)` method in `global.Go` so that TinyGo's runtime can be initialized prior to calling any of the WebAssembly module's exported functions. But there doesn't seem to be anything in `global.Go` indicating whether the runtime has been initialized. An easy workaround would be to create a wrapper object for `run` elsewhere in Javascript, but I feel like it would be useful to store a variable indicating the initialization status of the runtime in `global.Go` itself.

dgryski, 2022-05-19:
This seems pretty doable.

codefromthecrypt, 2022-09-07:
do any other compilers do this (rust, clang etc)? might be nice to see how.

dgryski, 2023-02-02:
If this ends up being part of https://github.com/tinygo-org/tinygo/issues/2735 then we could close this one.