tinygo-org/tinygo #2856: fatal: undefined symbol: syscall.Getpagesize

Theodoree, 2022-05-19:
env:

```
os:        debian 9
go:        1.18.1
tinygo:    0.23
llvm:      14.0.0

GOOS="linux"
GOARCH="amd64"
GOROOT="/usr/local/go"
GOPATH="/go"
GOCACHE="/root/.cache/tinygo"
CGO_ENABLED="1"
TINYGOROOT="/usr/local/lib/tinygo"
```

code:
```
package main
import (
     "fmt"
     "syscall"
)
func main() {
     fmt.Println(syscall.Getpagesize())
}
```

execute a command: `GOOS=linux GOARCH=amd64 tinygo build`
```
tinygo:ld.lld: error: undefined symbol: syscall.Getpagesize
>>> referenced by m.go:9 (/opt/repos/pkg/m.go:9)
>>>               /root/.cache/tinygo/thinlto/llvmcache-1F80CCA51D93267E9887A910A33CFCDC9CFE6A98:(runtime.run$1$gowrapper)
failed to run tool: ld.lld
error: failed to link /tmp/tinygo811094212/main: exit status 1
```
---
but darwin can be compiled normally, l don't  know way.

dankegel, 2022-05-19:
https://github.com/golang/go/blob/128279e5034ca29bad4320eef81a8abd5b40ea7e/src/syscall/syscall.go#L99 suggests it needs to be added to tinygo's runtime package on Linux.

dkegel-fastly, 2022-06-11:
Does https://github.com/tinygo-org/tinygo/pull/2899 work for you?

dkegel-fastly, 2022-06-12:
A fix has landed in the dev branch.

What app were you trying to build?

closed 2022-07-01 by deadprogram