tinygo-org/tinygo #2861: "tinygo test os" fails early on darwin m1 arm64

dkegel-fastly, 2022-05-20:
With current dev on my m1 mac, "tinygo test os" aborts early.

Doing "tinygo test -c os" and then single-stepping through os.test (and probably hello world), it seems to crash on the call to tinygo_swapTask:
```
...
(lldb) 
Process 13303 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = instruction step over
    frame #0: 0x0000000100006284 os.test`runtime.runMain + 212
os.test`runtime.runMain:
->  0x100006284 <+212>: bl     0x10002d800               ; tinygo_swapTask
    0x100006288 <+216>: str    xzr, [x23, #0x80]
    0x10000628c <+220>: b      0x10000624c               ; <+156>
    0x100006290 <+224>: ldp    x29, x30, [sp, #0x30]
Target 0: (os.test) stopped.
(lldb) ni
panic: runtime error: nil pointer dereference
```

aykevl, 2022-05-21:
Some questions:

 * Does this only happen for `tinygo test os` or also for other binaries?
 * This prints `panic: runtime error: nil pointer dereference`. Is that also what you get when you run the binary normally (outside of a debugger)?
 * Can you try setting a breakpoint on `runtime.nilpanic` and printing the stack trace when it is reached?
 * Is it still reproducible with `-opt=1`?

Not sure whether any of this works: it appears like DWARF debug information is broken on MacOS arm64.

dkegel-fastly, 2022-05-21:
It only happens for "tinygo test os"

It happens the same whether I run os.test inside or outside of a debugger.

Here's the backtrace right before the crash:
```
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 4.1
  * frame #0: 0x0000000100017a28 os.test`testing.tRunner + 4
    frame #1: 0x00000001000073f0 os.test`runtime.run$1$gowrapper + 4436
    frame #2: 0x000000010002d7fc os.test`tinygo_startTask at task_stack_arm64.S:28
```
Here's the backtrace after setting a breakpoint on runtime.nilpanic (which is tricky, you have to let it crash once, then set a breakpoint at the numerical address for some reason):
```
(lldb) b 0x0000000100004418
Breakpoint 2: where = os.test`runtime.nilPanic, address = 0x0000000100004418
(lldb) r
Process 14719 launched: '/Users/dkegel/src/tinygo-org/tinygo/os.test' (arm64)
Process 14719 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 2.1
    frame #0: 0x0000000100004418 os.test`runtime.nilPanic
os.test`runtime.nilPanic:
->  0x100004418 <+0>:  adrp   x0, 42
    0x10000441c <+4>:  mov    w1, #0x17
    0x100004420 <+8>:  add    x0, x0, #0x3bf
    0x100004424 <+12>: bl     0x100004094               ; runtime.runtimePanic
Target 0: (os.test) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 2.1
  * frame #0: 0x0000000100004418 os.test`runtime.nilPanic
    frame #1: 0x000000010000eda8 os.test`(*os.File).Write + 148
    frame #2: 0x0000000100025aac os.test`os_test.writeFile + 96
    frame #3: 0x00000001000258dc os.test`os_test.TestFstat + 80
```
... which is rather informative.  Evidently it gets further than I thought.
I'll see if I can track it down more.

aykevl, 2022-05-21:
Too bad debug information doesn't work. Otherwise debugging should be a lot more useful.
(I might be able to look into this)

dkegel-fastly, 2022-05-21:
Oh, and -opt 1 didn't seem to help.

aykevl, 2022-05-21:
> Too bad debug information doesn't work. Otherwise debugging should be a lot more useful.

Update: I think this is because of how macOS stores debug information. Essentially, they're not stored in the final binary but kept inside the .o file - which is removed by TinyGo after linking.
(And I now realize that I have debugged this issue before and came to the same conclusion).

dkegel-fastly, 2022-05-22:
It kinda looks like TestFstat() calls writeFile() which calls OpenFile(), which faithfully does everything it should, but crashes during the return to writeFile().  But I could be confused...

dkegel-fastly, 2022-05-22:
lldb's bt gives a more useful backtrace at -opt 0:
```
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
...
    frame #4: 0x000000010000b1f8 os.test`runtime.nilPanic + 20
    frame #5: 0x000000010000aa3c os.test`interface:{Error:func:{}{basic:string}}.Error$invoke + 596
    frame #6: 0x000000010004ed10 os.test`os_test.writeFile + 104
    frame #7: 0x000000010004ea30 os.test`os_test.TestFstat + 152
```

aykevl, 2022-05-22:
You could also try #2865. I can't easily test it myself, but it might make debug information available.

dkegel-fastly, 2022-05-22:
A strange discovery:
```
dkegel@P03DT9HJF7 tinygo % build/tinygo test -opt 0 -v -run 'TestReadAtOffset' os        
=== RUN   TestReadAtOffset
--- PASS: TestReadAtOffset (0.00s)
PASS
ok  	os	0.515s
dkegel@P03DT9HJF7 tinygo % build/tinygo test -opt 0 -v -run 'TestReadAtOffset|NOMATCH' os
panic: runtime error: nil pointer dereference
FAIL	os	0.536s
FAIL
```

dkegel-fastly, 2022-05-22:
Another strange behavior: It seems like rebuilding it generates a broken binary unless I remove it first.  Strange.

The strange -run 'junk|TestReadAtOffset' test failure can be reproduced with test -c:
```
% rm -f ./os.test
% build/tinygo test -opt 0  -c os

% ./os.test -test.v -test.run 'NOMATCH|TestReadAtOffset'
panic: runtime error: nil pointer dereference
zsh: abort      ./os.test -test.v -test.run 'NOMATCH|TestReadAtOffset'

% ./os.test -test.v -test.run 'NOMATCH|TestReadAtOffsetXXX'
testing: warning: no tests to run
PASS

% ./os.test -test.v -test.run 'TestReadAtOffset' 
=== RUN   TestReadAtOffset
--- PASS: TestReadAtOffset (0.00s)
PASS

% ./os.test -test.v -test.run 'NOMATCH|TestReadAtOffset' 
panic: runtime error: nil pointer dereference
zsh: abort      ./os.test -test.v -test.run 'NOMATCH|TestReadAtOffset'
```

And here's the backtrace, for completeness:
```
% lldb ./os.test -- -test.run 'TestReadAtOffset|NOMATCH'           
...
(lldb) r
Process 21521 launched: '/Users/dkegel/src/tinygo-org/tinygo/os.test' (arm64)
panic: runtime error: nil pointer dereference
Process 21521 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x000000019dc9cdb8 libsystem_kernel.dylib`__pthread_kill + 8
libsystem_kernel.dylib`__pthread_kill:
->  0x19dc9cdb8 <+8>:  b.lo   0x19dc9cdd8               ; <+40>
    0x19dc9cdbc <+12>: pacibsp 
    0x19dc9cdc0 <+16>: stp    x29, x30, [sp, #-0x10]!
    0x19dc9cdc4 <+20>: mov    x29, sp
Target 0: (os.test) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x000000019dc9cdb8 libsystem_kernel.dylib`__pthread_kill + 8
    frame #1: 0x000000019dcd1ee0 libsystem_pthread.dylib`pthread_kill + 288
    frame #2: 0x000000019dc0c340 libsystem_c.dylib`abort + 168
    frame #3: 0x0000000100009a84 os.test`runtime.runtimePanic + 52
    frame #4: 0x000000010000b1e4 os.test`runtime.nilPanic + 20
    frame #5: 0x000000010001ccd8 os.test`(*os.File).ReadAt + 452
    frame #6: 0x00000001000513e0 os.test`os_test.TestReadAtOffset + 232
    frame #7: 0x00000001000326b8 os.test`testing.tRunner + 112
    frame #8: 0x0000000100032608 os.test`(*testing.T).Run + 452
    frame #9: 0x0000000100032e34 os.test`testing.runTests$1 + 132
    frame #10: 0x00000001000326b8 os.test`testing.tRunner + 112
    frame #11: 0x0000000100032d80 os.test`testing.runTests + 200
    frame #12: 0x0000000100032aa8 os.test`(*testing.M).Run + 188
    frame #13: 0x0000000100056524 os.test`main.main + 76
    frame #14: 0x000000010000d52c os.test`runtime.run$1 + 12
    frame #15: 0x000000010000d33c os.test`runtime.run$1$gowrapper + 8
    frame #16: 0x00000001000565b0 os.test`tinygo_startTask at task_stack_arm64.S:28
```

aykevl, 2022-11-05:
`tinygo test os` now works on darwin/arm64.

deadprogram, 2023-02-12:
0.27.0 has now been released, so closing. Thank you!

closed 2023-02-12 by deadprogram