tinygo-org/tinygo #2866: "make test-corpus" hogs memory?

dkegel-fastly, 2022-05-22:
On my 12 GB linux box (gimme a break, I bought it in 2015) I run out of memory or time when running "make test-corpus"
unless I close chrome.   That last gigabyte or two of RAM is important, I guess.

Memory use seems to creep up slowly, hitting something like 15GB by the end of the test.

aykevl, 2022-05-22:
I think we're leaking a lot of memory. I've tried to fix this at least somewhat in https://github.com/tinygo-org/tinygo/pull/2785 but that PR broke things (and I haven't yet investigated why).

aykevl, 2022-05-29:
Looks like actually freeing memory helps quite a bit, memory consumption now seems to stay roughly flat while testing many packages with https://github.com/tinygo-org/tinygo/pull/2785 applied.

deadprogram, 2022-05-30:
@dkegel-fastly please retry this with the current `dev` branch now that #2785 has been merged.

dankegel, 2022-05-30:
It's so much better.  
RSS actually *goes down* as each test finishes.  
Peak RSS now 4.2GB for tinygo.test and 300MB for main.

It's been grinding on stat/distmv for 37 minutes, maybe I should run test-corpus-fast instead next time.

```
$ ps augxw | grep test.v
dank      182983  104 17.6 5418936 2151128 pts/0 Sl+  07:50  50:14 /tmp/go-build4260757586/b001/tinygo.test -test.paniconexit0 -test.v=true -test.timeout=1h0m0s -test.run=TestCorpus -corpus=testdata/corpus.yaml
dank      192555 94.5  2.3 1049220 289360 pts/0  R+   08:04  32:05 /tmp/tinygo4122828164/main -test.v
$ sudo gdb
(gdb) attach 192555
(runtime.gcBlock).findHead (b=991333) at /home/dank/src/tinygo/src/runtime/gc_conservative.go:124
124		for b.state() == blockStateTail {
(gdb) bt
#0  (runtime.gcBlock).findHead (b=991333) at /home/dank/src/tinygo/src/runtime/gc_conservative.go:124
#1  0x0000000000224f11 in runtime.startMark (root=<optimized out>) at /home/dank/src/tinygo/src/runtime/gc_conservative.go:496
#2  0x0000000000224b54 in runtime.finishMark () at /home/dank/src/tinygo/src/runtime/gc_conservative.go:538
#3  runtime.GC () at /home/dank/src/tinygo/src/runtime/gc_conservative.go:411
#4  runtime.alloc (size=40, layout=<optimized out>) at /home/dank/src/tinygo/src/runtime/gc_conservative.go:284
#5  0x000000000024f41f in gonum.org/v1/gonum/stat.Mahalanobis (x=..., y=..., chol=<optimized out>) at /home/dank/go/pkg/mod/gonum.org/v1/gonum@v0.11.0/stat/statmat.go:134
#6  0x000000000025c3d4 in gonum.org/v1/gonum/stat/distmv.normalLogProb (x=..., chol=0x7f43366d6e90, logSqrtDet=0.13501356860652997, mu=...)
    at /home/dank/go/pkg/mod/gonum.org/v1/gonum@v0.11.0/stat/distmv/normal.go:212
#7  (*gonum.org/v1/gonum/stat/distmv.Normal).LogProb (n=<optimized out>, x=...) at /home/dank/go/pkg/mod/gonum.org/v1/gonum@v0.11.0/stat/distmv/normal.go:186
#8  0x000000000026263e in gonum.org/v1/gonum/stat/distmv.bhattacharyyaSample (dim=2, samples=100000, l=..., r=...)
    at /home/dank/go/pkg/mod/gonum.org/v1/gonum@v0.11.0/stat/distmv/statdist_test.go:101
#9  0x000000000026248a in gonum.org/v1/gonum/stat/distmv.TestBhattacharyyaNormal (t=0x7f43366d6800)
    at /home/dank/go/pkg/mod/gonum.org/v1/gonum@v0.11.0/stat/distmv/statdist_test.go:44
```

dankegel, 2022-05-30:
"make test-corpus-fast" finishes in ten minutes on my 12GB i3-4130.
Booting with mem=6G to restrict memory to 6GB, and running with five chrome tabs open (gmail, facebook, twitter, github, and youtube playing a video) still finished in 10:38.

Rewinding back to before the patch, "make test-corpus-fast" used ~15GB swap space; peak RSS was 4.0GB, with occasional swapping; finished in 17:01.  Evidently all that swapping slowed it down.

dankegel, 2022-05-30:
The patch is great.  But even with, "make test-corpus" times out after one hour.

Main keeps running (so much for that timeout); after three hours, gdb says it is stuck at:
```
#0  0x0000000000224f96 in (runtime.gcBlock).state (b=148845) at /home/dank/src/tinygo/src/runtime/gc_conservative.go:150
#1  (runtime.gcBlock).findHead (b=148845) at /home/dank/src/tinygo/src/runtime/gc_conservative.go:124
#2  0x0000000000224f11 in runtime.startMark (root=<optimized out>) at /home/dank/src/tinygo/src/runtime/gc_conservative.go:496
#3  0x0000000000224b54 in runtime.finishMark () at /home/dank/src/tinygo/src/runtime/gc_conservative.go:538
#4  runtime.GC () at /home/dank/src/tinygo/src/runtime/gc_conservative.go:411
#5  runtime.alloc (size=40, layout=<optimized out>) at /home/dank/src/tinygo/src/runtime/gc_conservative.go:284
#6  0x000000000024f41f in gonum.org/v1/gonum/stat.Mahalanobis (x=..., y=..., chol=<optimized out>)
    at /home/dank/go/pkg/mod/gonum.org/v1/gonum@v0.11.0/stat/statmat.go:134
#7  0x000000000025c3d4 in gonum.org/v1/gonum/stat/distmv.normalLogProb (x=..., chol=0x7f01f232ce90, logSqrtDet=0.13501356860652997, mu=...)
    at /home/dank/go/pkg/mod/gonum.org/v1/gonum@v0.11.0/stat/distmv/normal.go:212
#8  (*gonum.org/v1/gonum/stat/distmv.Normal).LogProb (n=<optimized out>, x=...)
    at /home/dank/go/pkg/mod/gonum.org/v1/gonum@v0.11.0/stat/distmv/normal.go:186
#9  0x000000000026263e in gonum.org/v1/gonum/stat/distmv.bhattacharyyaSample (dim=2, samples=100000, l=..., r=...)
    at /home/dank/go/pkg/mod/gonum.org/v1/gonum@v0.11.0/stat/distmv/statdist_test.go:101
#10 0x000000000026248a in gonum.org/v1/gonum/stat/distmv.TestBhattacharyyaNormal (t=0x7f01f232c800)
    at /home/dank/go/pkg/mod/gonum.org/v1/gonum@v0.11.0/stat/distmv/statdist_test.go:44
```

I'll let it run overnight and see if it completes.

dankegel, 2022-05-30:
OK, this is messed up somehow.  Running the test on its own works fine:
```
$ time ~/src/tinygo/build/tinygo test -v -tags noasm gonum.org/v1/gonum/stat/distmv
PASS
ok  	gonum.org/v1/gonum/stat/distmv	13.104s

real	0m36.587s
user	0m39.975s
sys	0m0.774s
```
so why is it taking forever when part of "make test-corpus"?

It's like it's stuck in gc...?

dankegel, 2022-05-30:
The memory hogging behavior seems gone.  I'll file a new issue for the nontermination.

dkegel-fastly, 2022-05-31:
Filed apparent hang on some systems as https://github.com/tinygo-org/tinygo/issues/2883

closed 2022-07-01 by deadprogram