tinygo-org/tinygo #2877: panic when writing channel in  interrupt on RP2040

elbert-wong, 2022-05-30:
We got a "panic: runtime error: nil pointer dereference" when writing channel in interrupt handler, even if this channel is buffered.

With some investigation, we noticed that it's caused by the time.Sleep in go routine. Looks like it didn't yield to other go routine at the beginning of sleep.  My select to receive from channel(written by interrupt) didn't run until sleep finished.
With this code, if we send a pulse of 5Hz on interrupt pin, we got panic.  If we remove the LED blink go routine, this code works well. 

`package main

import (
	"machine"
	"runtime"
	"time"
)

const (
	pinLED = machine.LED
	pinIn = machine.GPIO12
)

var timeout = make(chan bool, 1)
var tstChan = make(chan int32, 3)

func pinConfig() {
	// Pin configuration

	pinIn.Configure(machine.PinConfig{Mode: machine.PinInputPullup})
	pinLED.Configure(machine.PinConfig{Mode: machine.PinOutput})
}

func main() {
	pinConfig()

	go func() {
		for {
			pinLED.High()
			time.Sleep(1000 * time.Millisecond)
			pinLED.Low()
			time.Sleep(1000 * time.Millisecond)
		}
	}()

	// go func() {
	// 	time.Sleep(3000 * time.Millisecond)
	// 	timeout <- true
	// }()

	pinIn.SetInterrupt(machine.PinFalling, pinFallHandler)
	for {
		select {
		case <-tstChan:
		case <-timeout:
			// not here
			// pinIn.SetInterrupt(machine.PinFalling, nil)
			goto DONE
		}
	}
DONE:
	for {
		runtime.Gosched()
	}
}

// 200ms trigger
func pinFallHandler(pin machine.Pin) {
	tstChan <- 1
}
`

======
Here I added some debug output on pins. I sent 2 pulses to interrupt pin every 1.2 second and we can see that  the channel was read every 1 second and got 2 elements(toggle to high and them toggle to low). As expected the channel should be read immediately when it's not empty. 

[image]

elbert-wong, 2022-05-30:
this issue is found on tinygo 0.22.0

dankegel, 2022-05-30:
Does it help to use a nonblocking channel write?

cf. https://stackoverflow.com/questions/48953236/how-to-implement-non-blocking-write-to-an-unbuffered-channel

kenbell, 2022-05-30:
@elbert-wong - can you try on the dev branch?  It includes this change: https://github.com/tinygo-org/tinygo/commit/8f40b107d93f7e3c98d1e724943f3f2370989ec9

That should mean that on RP2040, the scheduler gets a chance to run after each interrupt even if all go routines are sleeping.

elbert-wong, 2022-05-31:
> 

The channel writing was not blocked, since I used a buffered channel, but the channel was full (e.g. 5 writing per second by interrupt, and an 1 second sleep in a go routine). That means the channel reading loop have not run for at least 0.6 second. 

I think it's actually not a channel issue but a co-routine scheduler issue.

elbert-wong, 2022-05-31:
> @elbert-wong - can you try on the dev branch? It includes this change: [8f40b10](https://github.com/tinygo-org/tinygo/commit/8f40b107d93f7e3c98d1e724943f3f2370989ec9)
> 
> That should mean that on RP2040, the scheduler gets a chance to run after each interrupt even if all go routines are sleeping.

@kenbell Thank you!
Seems like you fixed this issue!!
Now the channel can be read out in about a couple of us, at the same time, a go routine is in sleep.
Thank you again!

closed 2022-07-01 by deadprogram