tinygo-org/tinygo #2883: dev's "make corpus-test" hangs on some systems?

dkegel-fastly, 2022-05-31:
I have had trouble with "make corpus-test" but not "make corpus-test-fast" seeming to livelock today on my dual-core linux system.
Adding RAM sometimes makes the livelock go away.
Getting a backtrace seems to always point to one test, and always involves runtime.GC.

I'll try to reproduce it on a different machine.  In the meantime, here's a representative log.

```
# Booted with mem=8G

$ uname -a
Linux thinky 5.15.0-33-generic #34-Ubuntu SMP Wed May 18 13:34:26 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

$ free
               total        used        free      shared  buff/cache   available
Mem:         7580984     2936240     2340128        4120     2304616     4379736
Swap:       28207096           0    28207096

$ cat /proc/cpuinfo | grep model | uniq
model name : Intel(R) Core(TM) i3-4130 CPU @ 3.40GHz

$ time make test-corpus

USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
dank        3433 97.8  3.8 1049220 289364 pts/0  R+   20:34  17:55 /tmp/tinygo3403064930/main -test.v

#0  0x0000000000224f91 in (runtime.gcBlock).state (b=1009786)
    at tinygo/src/runtime/gc_conservative.go:150
#1  (runtime.gcBlock).findHead (b=1009786)
    at tinygo/src/runtime/gc_conservative.go:124
#2  0x0000000000224f11 in runtime.startMark (root=<optimized out>)
    at tinygo/src/runtime/gc_conservative.go:496
#3  0x0000000000224b54 in runtime.finishMark ()
    at tinygo/src/runtime/gc_conservative.go:538
#4  runtime.GC () at tinygo/src/runtime/gc_conservative.go:411
#5  runtime.alloc (size=40, layout=<optimized out>)
    at tinygo/src/runtime/gc_conservative.go:284
#6  0x000000000024f41f in [gonum.org/v1/gonum/stat.Mahalanobis](http://gonum.org/v1/gonum/stat.Mahalanobis) (x=..., y=...,
    chol=<optimized out>)
    at go/pkg/mod/[gonum.org/v1/gonum@v0.11.0/stat/statmat.go:134](http://gonum.org/v1/gonum@v0.11.0/stat/statmat.go:134)
#7  0x000000000025c3d4 in [gonum.org/v1/gonum/stat/distmv.normalLogProb](http://gonum.org/v1/gonum/stat/distmv.normalLogProb) (x=...,
    chol=0x7f507c438e90, logSqrtDet=0.13501356860652997, mu=...)
    at go/pkg/mod/[gonum.org/v1/gonum@v0.11.0/stat/distmv/normal.go:212](http://gonum.org/v1/gonum@v0.11.0/stat/distmv/normal.go:212)
#8  (*[gonum.org/v1/gonum/stat/distmv.Normal).LogProb](http://gonum.org/v1/gonum/stat/distmv.Normal).LogProb) (n=<optimized out>, x=...)
    at go/pkg/mod/[gonum.org/v1/gonum@v0.11.0/stat/distmv/normal.go:186](http://gonum.org/v1/gonum@v0.11.0/stat/distmv/normal.go:186)
#9  0x000000000026263e in [gonum.org/v1/gonum/stat/distmv.bhattacharyyaSample](http://gonum.org/v1/gonum/stat/distmv.bhattacharyyaSample) (
    dim=2, samples=100000, l=..., r=...)
   onum@v0.11.0/stat/distmv/statdist_test.go:101
#10 0x000000000026248a in [gonum.org/v1/gonum/stat/distmv.TestBhattacharyyaNormal](http://gonum.org/v1/gonum/stat/distmv.TestBhattacharyyaNormal) (t=0x7f507c438800)
    at go/pkg/mod/[gonum.org/v1/gonum@v0.11.0/stat/distmv/statdist_test.go:44](http://gonum.org/v1/gonum@v0.11.0/stat/distmv/statdist_test.go:44)

=== CONT  TestCorpus/[gonum.org/v1/gonum(noasm-appengine)/lapack/gonum](http://gonum.org/v1/gonum(noasm-appengine)/lapack/gonum)
    main_test.go:516: ok   [gonum.org/v1/gonum/lapack/gonum](http://gonum.org/v1/gonum/lapack/gonum) 93.754s
^C^CFAIL [github.com/tinygo-org/tinygo](http://github.com/tinygo-org/tinygo) 1209.757s
make: *** [Makefile:366: test-corpus] Interrupt

real 20m18.441s
user 7m49.766s
sys 0m6.473s

# Minimized testdata/corpus.yaml down to:

- repo: [gonum.org/v1/gonum](http://gonum.org/v1/gonum)
  tags: noasm appengine
  subdirs:
  - pkg: blas/blas32
  - pkg: blas/cblas128
  - pkg: integrate
  - pkg: integrate/quad
  - pkg: internal/cmplx64
  - pkg: internal/testrand
  - pkg: lapack/gonum
    skipwasi: true
    slow: true
  - pkg: optimize/convex/lp
    skipwasi: true
  - pkg: optimize/functions
  - pkg: spatial/r2
  - pkg: spatial/r3
  - pkg: stat/distmat
  - pkg: stat/mds
  - pkg: stat/spatial
  - pkg: stat/distmv
    skipwasi: true
    slow: true
  - pkg: stat/samplemv
    skipwasi: true

```

aykevl, 2022-05-31:
Can you try whether this issue is still reproducible with https://github.com/tinygo-org/tinygo/compare/dev...gc-headroom? I'm suspecting it's constantly running the GC because it's almost, but not entirely, out of memory.
Disabling ASLR might make this issue more reproducible.

dkegel-fastly, 2022-05-31:
The problem seems to go away on the gc-headroom branch.

aykevl, 2022-05-31:
Thank you! I've made a PR (https://github.com/tinygo-org/tinygo/pull/2884) because I believe I've seen this issue pop up a few times before. And I think it's a change we need to make anyway.

dgryski, 2023-02-02:
Can we close this then?  We've had a few improvements to the gc (the headroom and precise GC fixes) land.

dkegel-fastly, 2023-02-02:
Let's call it done.

closed 2023-02-02 by dkegel-fastly