tinygo-org/tinygo #2887: Bad append() during initialization?

dkegel-fastly, 2022-06-01:
In a function called from a global variable initializer, I had to do replace an append with a make and six assignments:

```
var theStrings []string = getStrs()

func getStrs() []string {
...
... many lines deleted, the strings are somewhat long ...
...
-       strs := append([]string{}, str0, str1, str2, str3, str4, str5)
+       strs := make([]string, 6)
+       strs[0] = str0
+       strs[1] = str1
+       strs[2] = str2
+       strs[3] = str3
+       strs[4] = str4
+       strs[5] = str5
        return strs
}
```
else I ended up with the last five elements of strs[] being the empty string. 

This was with dev tinygo on macos, and it was the first time I'd tried compiling this code with tinygo.

I will try to come up with a working minimal reproducer...

dkegel-fastly, 2022-06-02:
Also fails:
-  tinygo 0.23 on mac from homebrew

dgryski, 2022-06-07:
I'll take a look at this.

dgryski, 2022-06-17:
I'm now beginning to suspect this is a problem wth `interp`.

radkomih, 2023-07-07:
@dgryski could you elaborate on how to disable the `interp`? I am experiencing a similar bug in my custom target based on the latest release 0.28.1.

dkegel-fastly, 2023-07-07:
you could try -timeout 0, or hack this line in builder/build.go to always take the error path:
```
                                err := interp.RunFunc(pkgInit, config.Options.InterpTimeout, config.DumpSSA())
                                if err != nil {
                                        return err
                                }
```

dgryski, 2024-10-07:
This appears to be fixed now.  I'm tempted to bisect and see what the commit was that caused the problem to vanish.

dgryski, 2024-10-07:
`git bisect` shows https://github.com/tinygo-org/tinygo/commit/417a26d20c23ea0aaa8f95325bb30172b3e3e9e9 .  Tagging this as "next release".

dgryski, 2024-10-07:
Actually this is in v0.33 already.  Closing.

closed 2024-10-07 by dgryski