tinygo-org/tinygo #2890: Wrong signature of imported function

vlkv, 2022-06-02:
Hi, here is my "hello wasm world". Here I am trying to import Add function:
```
// main.go 
package main

//export Sum
func Sum(a int32, b int32) int32 {
	return Add(a, b)
}

func Add(a, b int32) int32

func main() {
}
```

I compile it to wasm with a command:
```
tinygo build -o main.wasm -target wasm ./main.go
```
Then I inspect the module with 
```
$ wasmer inspect main.wasm 
Type: wasm
Size: 56.0 KB
Imports:
  Functions:
    "wasi_snapshot_preview1"."fd_write": [I32, I32, I32, I32] -> [I32]
    "env"."main.Add": [I32, I32, I32] -> [I32]
  Memories:
  Tables:
  Globals:
Exports:
  Functions:
    "malloc": [I32] -> [I32]
    "free": [I32] -> []
    "calloc": [I32, I32] -> [I32]
    "realloc": [I32, I32] -> [I32]
    "_start": [] -> []
    "resume": [] -> []
    "go_scheduler": [] -> []
    "Sum": [I32, I32] -> [I32]
    "asyncify_start_unwind": [I32] -> []
    "asyncify_stop_unwind": [] -> []
    "asyncify_start_rewind": [I32] -> []
    "asyncify_stop_rewind": [] -> []
    "asyncify_get_state": [] -> [I32]
  Memories:
    "memory": not shared (2 pages..)
  Tables:
  Globals:
```

Why the signature of Add function has THREE parameters? `"env"."main.Add": [I32, I32, I32] -> [I32]`
ALSO: If I add `//export Add` comment
```
//export Add
func Add(a, b int32) int32
```
then the signature after compilation looks fine. If this is by design, then can you please explain what is going on here. Otherwise, It looks like a bug to me. 

Thanks.

aykevl, 2022-06-02:
This is working as intended. Exported functions have a defined calling convention but non-exported functions do not. You need to add `//export` to make a function available externally (even for imported functions).
The extra parameter for non-exported functions is an implementation detail and can change at any time. At the moment, it is used for passing extra data to closures.

closed 2022-06-02 by aykevl

vlkv, 2022-06-02:
OK, thank you very much.