tinygo-org/tinygo #2892: Adds support for `./...` syntax

jcchavezs, 2022-06-03:
Running `tinygo test ./...` does not properly resolve all the packages recursively.

dkegel-fastly, 2022-06-03:
Seems to resolve recursively here.  Can you give a real-world example?  e.g.
```
$ git clone https://go.googlesource.com/image
$ cd image
$ tinygo test ./...
# golang.org/x/image/font/inconsolata
font/inconsolata/bold8x16.go:8:2: could not import golang.org/x/image/font/basicfont (Config.Importer.Import(golang.org/x/image/font/basicfont) returned nil but no error)
FAIL
```
(That's just an example of the kind of detail that would be great in this bug report)

jcchavezs, 2022-06-03:
In this very same repository, you can run

```
➜  tinygo git:(fixes_three_dots) ✗ tinygo test ./tests/testing/... 
error: Linking globals named 'main.tests': symbol multiply defined!
```
and after the change you get
```
➜  tinygo git:(fixes_three_dots) ✗ ~/Workspace/gource/bin/tinygo test ./tests/testing/... 
ld.lld: error: undefined symbol: _notARealFunction
>>> referenced by /var/folders/s7/qmqrxc894d55fymyt4x506300000gn/T/tinygo1047012886/main.o
error: failed to link /var/folders/s7/qmqrxc894d55fymyt4x506300000gn/T/tinygo1047012886/main: exit status 1
--- FAIL: TestFail (0.00s)
    fail
FAIL
FAIL    github.com/tinygo-org/tinygo/tests/testing/fail 0.326s
?       github.com/tinygo-org/tinygo/tests/testing/nothing      [no test files]
ok      github.com/tinygo-org/tinygo/tests/testing/pass 0.176s
FAIL
````

But all this started in the repository github.com/corazawaf/coraza where running `tinygo test ./...` only tested first package `github.com/corazawaf/coraza/v2`.

dkegel-fastly, 2022-06-04:
I think it stops at the first failure.  It'll be a while before coraza builds with tinygo, I think.

Here's a PR to add a smoke test for "tinygo test ./...": https://github.com/tinygo-org/tinygo/pull/2894

I think your branch currently fails that smoke test, FWIW.

jcchavezs, 2022-06-04:
> I think it stops at the first failure. It'll be a while before coraza builds with tinygo, I think.

I am working on making it support tinygo, see https://github.com/corazawaf/coraza/pull/254

> I think your branch currently fails that smoke test, FWIW.

I just left you a comment, I think the test has a bug, see https://github.com/tinygo-org/tinygo/pull/2894#discussion_r889616906

dkegel-fastly, 2022-06-04:
That explains a lot :-). Glad to see your work!

Thank you for reviewing my test; I checked your theory, seems it is incorrect; it would have been correct had I been quoting the result of the wc.

PR #2885 (2022-06-11): fixes `./...` syntax in `tinygo test`
Adds proper support for `./...` syntax which was broken or maybe never intended to be supported and it only somehow worked because we use `go list` underneath. Allowing to use `./...` is actually a good experience in `go test`.

For example, running the tinygo@0.23.0 like:

```
$ tinygo test ./tests/testing/threedots/...
ok      github.com/tinygo-org/tinygo/tests/testing/threedots/pass       0.106s
```

whereas `./tests/testing/threedots` has two packages.

Closes #2892
closed 2022-07-01 by deadprogram