tinygo-org/tinygo #2895: I2C slave mode support ?

NV4RE, 2022-06-05:
Hello, I'm recently trying [tinygo](https://github.com/tinygo-org/tinygo), and quite happy from it, but I face some problem that I need to have i2c slave (server) but I did not find any document about it, so anyone could suggest me where to find the example or ducument about imprementing i2c slave on tinygo, if it dose not support please give me some advice how to workaround with that.

My use-case is I'm using `bluepill` as main interface controller to handle the gpio, and devices such as (DS3231, ili9341, battery volatage monitor, and ext-watchdog timer to reset main computer, dipswitchs and buttons), the reason I did not handle those in main machine because I need flexibility to change the main computing to other platform such as (RPi, OPi, nanopi) with out hardware limitation E.g. (Some Orange pi did not map the gpio to complatible with RPi standard)

My hardware design contains 3 part
1. Main computing (Just a SBC that runing linux with custom application) `Writen in standard go`
2. Main interface board (contain standard requirement such as RTC, screen, configuration DIP switch, D-pad, Backup Battery)  `Writen in tinygo`
3. Application specific board (specially made for each use-case e.g. `board with RS-422`, `board with multiple dry-contracts and relay`, `board with Lora module`) `Writen in tinygo`

That is the reason why I need I2C slave (I need way to comunication to multiple slave)

soypat, 2022-06-05:
If I understand correctly your setup must do the following: 
1. Your SBC (linux) sends commands to the main interface board. You want to send commands over I2C if I understood correctly.
2. Main interface board interprets these commands and performs tasks such as
   - Interfaciung via RS-422 with a application specific board
   - Activating GPIOs
   - Doing some SPI transaction or other protocol
   
If this is the case I'm afraid I have no experience setting up I2C for slave mode or any idea of how that would work. What I can tell you is I must do something similar at work and I plan to use UART for this purpose since it is much more flexible than i2c. If your SBC has a USB port you can buy a UART to ttl converter (commonly known as UART to USB converter) and have your linux machine interface on the corresponding `/dev/ttyUSBx` file interface. After having this set up a protocol must be developed- It is my understanding something of the sort of [UAVCAN/CYPHAL](https://opencyphal.org/specification/Cyphal_Specification.pdf) or similar could be used.

NV4RE, 2022-06-08:
Hello @soypat, thank you for your responsive respond, [CYPHAL](https://opencyphal.org/specification/Cyphal_Specification.pdf) seem to fit into my use-case.

closed 2022-06-08 by NV4RE