tinygo-org/tinygo #2907: Warnings are printed when running tinygo run or tinygo build on Windows

sago35, 2022-06-13:
```
$ tinygo run ./src/examples/serial
ld.lld: warning: duplicate /export option: hypot
ld.lld: warning: duplicate /export option: nextafter
hello world!
hello world!
hello world!
hello world!

$ tinygo version
tinygo version 0.24.0-dev-8754f64f windows/amd64 (using go version go1.18.1 and LLVM version 14.0.0)
```

The same message is displayed in CI.

https://github.com/tinygo-org/tinygo/runs/6846276428?check_suite_focus=true#step:18:493

```
 GOOS=windows GOARCH=amd64 D:/a/tinygo/tinygo/build/tinygo build -size short -o test.exe   ./testdata/cgo
ld.lld: warning: duplicate /export option: hypot
ld.lld: warning: duplicate /export option: nextafter
   code    data     bss |   flash     ram
  21663     512       0 |   22175     512
```

aykevl, 2022-06-16:
Yes, I'm aware of those. Those flags aren't actually passed to the linker but these symbols are part of mingw-w64.

dulanov, 2022-07-19:
FYI: https://sourceforge.net/p/mingw-w64/mingw-w64/merge-requests/20/

aykevl, 2022-08-28:
This is fixed upstream: https://github.com/mingw-w64/mingw-w64/commit/8526cb618269440a94810b94b77f8bd48c5c3396
We could update now, or wait until the next mingw-w64 release. I think it's fine to wait, the message is annoying but harmless.

deadprogram, 2022-09-11:
> I think it's fine to wait, the message is annoying but harmless.

Agreed, let's do nothing here. :smiley_cat:

davecheney, 2026-10-24:
Fixed by https://github.com/tinygo-org/tinygo/commit/7aca814954e83d14a62b110a70c9aaeb32fea999, which updates mingw-w64 to the upstream fix for the duplicate `hypot` and `nextafter` exports.

This small program cross-compiles for Windows without either warning:

```go
package main

func main() {
    println("hello world!")
}
```

```sh
GOOS=windows GOARCH=amd64 tinygo build -o check.exe main.go
```

Checked with TinyGo 0.44.0-dev-eae3f875 and Go 1.27.3. Please open a new issue if you encounter further problems.

closed 2026-10-24 by davecheney