tinygo-org/tinygo #2922: []byte{} leads to unknown arg type: []byte

Rail-dex, 2022-06-20:
Trying to use a byte array throws a compilation error : 

unknown arg type: []byte

Doesnt matter if self written or in any predesigned struct like bytes.buffer{}

tinygo version 0.23.0 windows/amd64 (using go version go1.18.3 and LLVM version 14.0.0)
MCU used: RPI PICO

aykevl, 2022-06-21:
What is the code you're compiling?

Rail-dex, 2022-06-21:
package main

	func main() {

		buffer := []byte{'H', 'e', 'l', 'l', 'o', ' ', 'W', 'o', 'r', 'l', 'd'}

		print(buffer)

	}

dgryski, 2022-06-22:
TinyGo's `print` builtins don't supprt `[]byte`: https://github.com/tinygo-org/tinygo/blob/release/compiler/compiler.go#L1304

Try `print(string(buffer))` instead.

deadprogram, 2023-05-22:
This was addressed by #2925 quite a while ago, so closing. Thanks again everyone!

PR #2925 (2022-07-13): Add support for printing slices via print/println
Fixes #2922.
closed 2023-05-22 by deadprogram