tinygo-org/tinygo #2938: RP2040 sometimes does not start up

sago35, 2022-06-28:
As shown in the following image, pressing the RESET button while Blinky1 is running may not start it.
It always seems to work immediately after `tinygo flash`.
It may not work after pressing RESET or turning the power back on.
There may be a problem with the initialization code.

Reproduced with TinyGo 0.23 and dev branch.

[image]

kenbell, 2022-07-04:
@sago35 I've noticed some slight weirdness on reset sometimes - putting a sleep at the start of main fixed it for me.  Wonder if it's just a timing thing?

sago35, 2022-07-05:
After making the changes in #2960, the problem has been completely resolved.

From @kenbell's advice, I was able to check pico and notice it.
Thank you very much.

closed 2022-08-04 by deadprogram