tinygo-org/tinygo #2939: how to pass js array to wasm ?

826327700, 2022-06-29:
who can give me a exmaple ! thanks so much!

vlkv, 2022-07-18:
> I believe you must pass it as a string.
> So on the javascript side, you would do something like
> `myWasmFunction(JSON.stringify(myArray))`
Can you provide a full example about how to pass a string to wasm module? I thought this should be accomplished by
- manually allocating memory for the string inside wasm
- copying the string to the allocated memory
- passing to `myWasmFunction` address of the string start and string length.
If we can do all the same with TinyGo in a different, simpler way, please, tell us how to do it.

codefromthecrypt, 2022-09-07:
Good news. Thanks to @trivigy, there's an example for this.

Follow the directions here, except `make slices` instead of `make main` https://github.com/tinygo-org/tinygo/blob/release/src/examples/wasm/README.md

I verified this just now. https://github.com/tinygo-org/tinygo/blob/release/src/examples/wasm/slices/wasm.go

jiangbo66666, 2024-04-29:
> Good news. Thanks to @trivigy, there's an example for this.
>
> Follow the directions here, except `make slices` instead of `make main` [...]

It means we have to pass Array with the help of "syscall/js" ? or pass it as a string . i want know which will faster ?

davecheney, 2026-10-27:
The array interop example is in `src/examples/wasm/slices`. Closing this as answered. Performance depends on the data and how it is passed; compare the approaches with a benchmark for your use case.

closed 2026-10-27 by davecheney