tinygo-org/tinygo #2951: Call parameter type does not match function signature! when method input parameter is a generic type

yoursunny, 2022-07-01:
**main.go**

```go
package main

type C[V any] struct{}

func (c *C[V]) F(v N[V]) {}

type N[V any] struct{}

func main() {
	var c C[int]
	var n N[int]
	c.F(n)
}
```

Commands and error message:

```bash
yoursunny@localhost:~/snippet$ tinygo version
tinygo version 0.24.0 linux/amd64 (using go version go1.18.3 and LLVM version 14.0.0)

yoursunny@localhost:~/snippet$ tinygo build -target=wasm main.go
Call parameter type does not match function signature!
  %0 = bitcast i8* %c to %main.C*, !dbg !15
 %main.C.0*  call void @"(*main.C[int]).F[int]"(%main.C* %0, i8* undef), !dbg !17
error: verification error after compiling package command-line-arguments
```

nik0sc, 2022-07-15:
I faced a similar problem when compiling some code that uses generics. What I found is that this error occurs when there is at least one pointer argument to the method (including the receiver). So in the example code, if the method F has a value receiver `func (c C[V]) F(v N[V]) {}` the code compiles without errors.

aykevl, 2022-07-15:
See: https://github.com/tinygo-org/tinygo/pull/2998

deadprogram, 2023-08-26:
Closing since this is part of release 0.29.0 thank you!

closed 2023-08-26 by deadprogram