tinygo-org/tinygo #2955: cgo: get wrong int value from c struct and panic with "signal: segmentation fault"

leizongmin, 2022-07-03:
Here is an example code:

```go
package main

/*
typedef struct S {
  	int a;
	int b;
} S;

S test() {
  	S s;
  	s.a = 123;
	s.b = 456;
	return s;
}
*/
import "C"
import (
	"fmt"
)

func main() {
	C.test()
	s := C.test()
	fmt.Println(s.a, s.b)
	fmt.Println(int(s.a), int(s.b))
	fmt.Println(s)
}
```

run with `go run main.go` get the following output:

```
123 456
123 456
{123 456}
```

run with `tinygo run main.go` get the following output:

```
123 461824000
123 461824000
panic: (reflect.Value).Interface: unexported
error: failed to run compiled binary /var/folders/x2/9mjt7vw565b6n059lxlsh3900000gn/T/tinygo3450304379/main: signal: abort trap
```

aykevl, 2022-09-02:
Actually, this was probably #3033 which I've fixed in #3111.

aykevl, 2022-09-02:
No, _actually_ it was both issues. The panic has been fixed in #3111, but it still returns the wrong value.