tinygo-org/tinygo #2958: LLVM crash

kenbell, 2022-07-04:
I'm getting this crash from LLVM:
```
LLVM ERROR: underestimated function size
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Running pass 'Function Pass Manager' on module '/tmp/tinygo4269388313/main.o'.
1.      Running pass 'ARM constant island placement and branch shortening pass' on function '@main.main'
 #0 0x00007f726aed8d01 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xe3fd01)
 #1 0x00007f726aed6a3e llvm::sys::RunSignalHandlers() (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xe3da3e)
 #2 0x00007f726aed9236 (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xe40236)
 #3 0x00007f7269b80520 (/lib/x86_64-linux-gnu/libc.so.6+0x42520)
 #4 0x00007f7269bd4a7c __pthread_kill_implementation ./nptl/./nptl/pthread_kill.c:44:76
 #5 0x00007f7269bd4a7c __pthread_kill_internal ./nptl/./nptl/pthread_kill.c:78:10
 #6 0x00007f7269bd4a7c pthread_kill ./nptl/./nptl/pthread_kill.c:89:10
 #7 0x00007f7269b80476 gsignal ./signal/../sysdeps/posix/raise.c:27:6
 #8 0x00007f7269b667f3 abort ./stdlib/./stdlib/abort.c:81:7
 #9 0x00007f726ae13723 (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xd7a723)
#10 0x00007f726ae13556 (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xd7a556)
#11 0x00007f726ce120cb (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0x2d790cb)
#12 0x00007f726b25f2be llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0x11c62be)
#13 0x00007f726b0133c0 llvm::FPPassManager::runOnFunction(llvm::Function&) (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xf7a3c0)
#14 0x00007f726b01a9b3 llvm::FPPassManager::runOnModule(llvm::Module&) (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xf819b3)
#15 0x00007f726b013f66 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xf7af66)
#16 0x00007f726c3feaff (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0x2365aff)
#17 0x00007f726c3ffb48 (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0x2366b48)
#18 0x00007f726c3ff99a llvm::lto::thinBackend(llvm::lto::Config const&, unsigned int, std::function<llvm::Expected<std::unique_ptr<llvm::CachedFileStream, std::default_delete<llvm::CachedFileStream> > > (unsigned int)>, llvm::Module&, llvm::ModuleSummaryIndex const&, llvm::StringMap<std::unordered_set<unsigned long, std::hash<unsigned long>, std::equal_to<unsigned long>, std::allocator<unsigned long> >, llvm::MallocAllocator> const&, llvm::DenseMap<unsigned long, llvm::GlobalValueSummary*, llvm::DenseMapInfo<unsigned long, void>, llvm::detail::DenseMapPair<unsigned long, llvm::GlobalValueSummary*> > const&, llvm::MapVector<llvm::StringRef, llvm::BitcodeModule, llvm::DenseMap<llvm::StringRef, unsigned int, llvm::DenseMapInfo<llvm::StringRef, void>, llvm::detail::DenseMapPair<llvm::StringRef, unsigned int> >, std::vector<std::pair<llvm::StringRef, llvm::BitcodeModule>, std::allocator<std::pair<llvm::StringRef, llvm::BitcodeModule> > > >*, std::vector<unsigned char, std::allocator<unsigned char> > const&) (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0x236699a)
#19 0x00007f726c3f90e6 (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0x23600e6)
#20 0x00007f726c3f8d39 (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0x235fd39)
#21 0x00007f726b362d12 (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0x12c9d12)
#22 0x00007f726ae7b32b (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xde232b)
#23 0x00007f7269bd2b43 start_thread ./nptl/./nptl/pthread_create.c:442:8
#24 0x00007f7269c64a00 ./misc/../sysdeps/unix/sysv/linux/x86_64/clone3.S:83:0
error: failed to link /tmp/tinygo4269388313/main: signal: aborted (core dumped)
```

It looks like a genuine LLVM issue?

The error seems to relate to this LLVM code: https://github.com/llvm-mirror/llvm/blob/master/lib/Target/ARM/ARMConstantIslandPass.cpp#L1628
The change that introduced the error is: https://github.com/llvm-mirror/llvm/commit/aae431a7be048672cc98ce3bdb2268b681ab7201

I'm going to see if i can find a smaller repro case.  It's currently being provoked by example code for a LoRa driver I'm writing.  Randomly adding/removing code makes it go away, so it's a bit tricky to track down.

kenbell, 2022-07-04:
This is the LLVM IR for main.main that is crashing...
```
; Function Attrs: minsize nounwind optsize
define internal void @main.main(i8* %context) #0 !dbg !167697 {
entry:
  %complit12 = alloca %"tinygo.org/x/drivers/sx127x.Config", align 4, !dbg !167736
  %complit9 = alloca %machine.SPIConfig, align 4, !dbg !167736
  %complit = alloca %"runtime/volatile.Register8", align 4, !dbg !167736
  %.fca.0.gep = getelementptr inbounds %"runtime/volatile.Register8", %"runtime/volatile.Register8"* %complit, i32 0, i32 0, !dbg !167736
  store i8 0, i8* %.fca.0.gep, align 1, !dbg !167736
  br i1 false, label %store.throw, label %store.next, !dbg !167737

store.next:                                       ; preds = %entry
  store i8 0, i8* %.fca.0.gep, align 1, !dbg !167737
  call void @"(machine.Pin).Configure"(i8 13, i8 0, i8* undef), !dbg !167738
  call void @"(machine.Pin).Set"(i8 13, i1 false, i8* undef), !dbg !167739
  call void @time.Sleep(i64 1000000000, i8* undef), !dbg !167740
  %0 = load %machine.SPI*, %machine.SPI** @machine.SPI0, align 4, !dbg !167741
  %1 = icmp eq %machine.SPI* %0, null, !dbg !167742
  br i1 %1, label %deref.throw, label %deref.next, !dbg !167742

deref.next:                                       ; preds = %store.next
  %2 = load %machine.SPI, %machine.SPI* %0, align 4, !dbg !167742
  %.fca.0.gep20 = getelementptr inbounds %machine.SPIConfig, %machine.SPIConfig* %complit9, i32 0, i32 0, !dbg !167736
  store i32 0, i32* %.fca.0.gep20, align 4, !dbg !167736
  %.fca.1.gep = getelementptr inbounds %machine.SPIConfig, %machine.SPIConfig* %complit9, i32 0, i32 1, !dbg !167736
  store i1 false, i1* %.fca.1.gep, align 4, !dbg !167736
  %.fca.2.gep = getelementptr inbounds %machine.SPIConfig, %machine.SPIConfig* %complit9, i32 0, i32 2, !dbg !167736
  store i8 0, i8* %.fca.2.gep, align 1, !dbg !167736
  %.fca.3.gep = getelementptr inbounds %machine.SPIConfig, %machine.SPIConfig* %complit9, i32 0, i32 3, !dbg !167736
  store i8 0, i8* %.fca.3.gep, align 2, !dbg !167736
  %.fca.4.gep = getelementptr inbounds %machine.SPIConfig, %machine.SPIConfig* %complit9, i32 0, i32 4, !dbg !167736
  store i8 0, i8* %.fca.4.gep, align 1, !dbg !167736
  %.fca.5.gep = getelementptr inbounds %machine.SPIConfig, %machine.SPIConfig* %complit9, i32 0, i32 5, !dbg !167736
  store i8 0, i8* %.fca.5.gep, align 4, !dbg !167736
  %.fca.6.gep = getelementptr inbounds %machine.SPIConfig, %machine.SPIConfig* %complit9, i32 0, i32 6, !dbg !167736
  store i8 0, i8* %.fca.6.gep, align 1, !dbg !167736
  br i1 false, label %store.throw10, label %store.next11, !dbg !167743

store.next11:                                     ; preds = %deref.next
  store i32 1000000, i32* %.fca.0.gep20, align 4, !dbg !167743
  %.fca.1.load = load i1, i1* %.fca.1.gep, align 4, !dbg !167744
  %.fca.1.insert = insertvalue %machine.SPIConfig { i32 1000000, i1 poison, i8 poison, i8 poison, i8 poison, i8 poison, i8 poison }, i1 %.fca.1.load, 1, !dbg !167744
  %.fca.2.load = load i8, i8* %.fca.2.gep, align 1, !dbg !167744
  %.fca.2.insert = insertvalue %machine.SPIConfig %.fca.1.insert, i8 %.fca.2.load, 2, !dbg !167744
  %.fca.3.load = load i8, i8* %.fca.3.gep, align 2, !dbg !167744
  %.fca.3.insert = insertvalue %machine.SPIConfig %.fca.2.insert, i8 %.fca.3.load, 3, !dbg !167744
  %.fca.4.load = load i8, i8* %.fca.4.gep, align 1, !dbg !167744
  %.fca.4.insert = insertvalue %machine.SPIConfig %.fca.3.insert, i8 %.fca.4.load, 4, !dbg !167744
  %.fca.5.load = load i8, i8* %.fca.5.gep, align 4, !dbg !167744
  %.fca.5.insert = insertvalue %machine.SPIConfig %.fca.4.insert, i8 %.fca.5.load, 5, !dbg !167744
  %.fca.6.load = load i8, i8* %.fca.6.gep, align 1, !dbg !167744
  %.fca.6.insert = insertvalue %machine.SPIConfig %.fca.5.insert, i8 %.fca.6.load, 6, !dbg !167744
  %3 = extractvalue %machine.SPI %2, 0, !dbg !167745
  %4 = call %runtime._interface @"(machine.SPI).Configure"(%"device/rp.SPI0_Type"* %3, %machine.SPIConfig %.fca.6.insert, i8* undef), !dbg !167745
  %modem = call i8* @runtime.alloc(i32 16, i8* inttoptr (i32 649 to i8*), i8* undef), !dbg !167746
  %5 = bitcast i8* %modem to %"tinygo.org/x/drivers/sx127x.Device"*, !dbg !167746
  %6 = load %machine.SPI*, %machine.SPI** @machine.SPI0, align 4, !dbg !167747
  %pack.ptr = bitcast %machine.SPI* %6 to i8*, !dbg !167747
  %7 = insertvalue %runtime._interface { i32 ptrtoint (%runtime.typecodeID* @"reflect/types.type:pointer:named:machine.SPI" to i32), i8* undef }, i8* %pack.ptr, 1, !dbg !167747
  %8 = extractvalue %runtime._interface %7, 0, !dbg !167748
  %9 = call %"tinygo.org/x/drivers/sx127x.Device" @"tinygo.org/x/drivers/sx127x.New"(i32 %8, i8* %pack.ptr, i8 28, i8 29, i8 6, i8* undef), !dbg !167748
  store %"tinygo.org/x/drivers/sx127x.Device" %9, %"tinygo.org/x/drivers/sx127x.Device"* %5, align 4, !dbg !167746
  call void @llvm.dbg.value(metadata %"tinygo.org/x/drivers/sx127x.Device" %9, metadata !167700, metadata !DIExpression()), !dbg !167746
  %.fca.0.gep30 = getelementptr inbounds %"tinygo.org/x/drivers/sx127x.Config", %"tinygo.org/x/drivers/sx127x.Config"* %complit12, i32 0, i32 0, !dbg !167736
  store i32 0, i32* %.fca.0.gep30, align 4, !dbg !167736
  %.fca.1.gep31 = getelementptr inbounds %"tinygo.org/x/drivers/sx127x.Config", %"tinygo.org/x/drivers/sx127x.Config"* %complit12, i32 0, i32 1, !dbg !167736
  store i8 0, i8* %.fca.1.gep31, align 4, !dbg !167736
[... cut ...]
  unreachable, !dbg !167743

store.throw13:                                    ; preds = %store.next11
  call void @runtime.nilPanic(i8* undef), !dbg !167749
  unreachable, !dbg !167749

store.throw15:                                    ; preds = %store.next14
  call void @runtime.nilPanic(i8* undef), !dbg !167750
  unreachable, !dbg !167750

lookup.throw:                                     ; preds = %for.body7
  call void @runtime.lookupPanic(i8* undef), !dbg !167791
  unreachable, !dbg !167791
}

```

tonygilkerson, 2022-08-22:
I am getting this error right after upgrading to go v1.19 I am on a mac.  What commands can I run to provide the proper information for this ticket?

```
tinygo build -target=pico
LLVM ERROR: underestimated function size
error: failed to link /var/folders/pt/rkqfz9js2kvf9cvkk5m052wm0000gp/T/tinygo3160594387/main: signal: abort trap
```

aykevl, 2022-08-23:
If you can provide the code so we can reproduce this that would be great!

roman-dvorak, 2022-11-17:
Hi, I am also falling into this error. 

```
$ tinygo flash -target=pico main.go -v
LLVM ERROR: underestimated function size
error: failed to link /tmp/tinygo2959922757/main: signal: aborted (core dumped)
```
It happens when I try to convert int16 value to float value.

```
  d.WriteByte(0xBE)
  var data [9]byte

  for i := 0; i < 8; i++ {
    data[i] = d.ReadByte()
  }

  var temp int16

  temp |= int16(data[0])
  temp |= (int16(data[1])&0x0f) << 8

  ftemp := float32(temp)/16.0
  println("RAWTEMP", temp, ftemp)
```

After deleting the float32 function from the code, compilation will occur without error.

I will be happy to provide more details.

aykevl, 2022-11-17:
> I will be happy to provide more details.

Yes, please provide a full reproducer. That way we can actually see what's going wrong.

- source code
- the command that you used to compile the binary
- `tinygo version`

deadprogram, 2023-08-18:
@kenbell is this still an issue with the latest `dev` branch?

scottfeldman, 2024-01-04:
I'm hitting this LLVM error with dev branch.  Seems 100% reproducible.  Here's what I did:

Modified drivers:examples/net/tlsclient/main.go with this patch

```
sfeldma@nuc:~/work/drivers$ git diff examples/
diff --git a/examples/net/tlsclient/main.go b/examples/net/tlsclient/main.go
index 186fdd5..cd0a1b6 100644
--- a/examples/net/tlsclient/main.go
+++ b/examples/net/tlsclient/main.go
@@ -28,10 +28,12 @@ var (
        ssid string
        pass string
        // HTTPS server address to hit with a GET / request
-       address string = "httpbin.org:443"
+       address  string = "httpbin.org:443"
+       address2 string = "192.168.1.213:8080"
 )
 
 var conn net.Conn
+var conn2 net.Conn
 
 // Wait for user to open serial console
 func waitSerial() {
@@ -70,6 +72,18 @@ func dialConnection() {
        println("Connected!\r")
 }
 
+func dialConnection2() {
+       var err error
+
+       println("\r\n---------------\r\nDialing TCP connection2")
+       conn2, err = net.Dial("tcp", address2)
+       for ; err != nil; conn2, err = net.Dial("tcp", address2) {
+               println("Connection failed:", err.Error())
+               time.Sleep(5 * time.Second)
+       }
+       println("Connected!\r")
+}
+
 func makeRequest() {
        print("Sending HTTPS request...")
        w := bufio.NewWriter(conn)
@@ -97,6 +111,7 @@ func main() {
 
        for i := 0; ; i++ {
                dialConnection()
+               dialConnection2()
                makeRequest()
                readResponse()
                closeConnection()
```

And try to build it:

```
sfeldma@nuc:~/work/drivers$ tinygo build -target nano-rp2040 -stack-size 4kb -ldflags="-X 'main.ssid=test' -X 'main.pass=testtest'" ./examples/net/tlsclient/
LLVM ERROR: underestimated function size
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Running pass 'Function Pass Manager' on module '/tmp/tinygo3956045343/main.o'.
1.      Running pass 'ARM constant island placement and branch shortening pass' on function '@main.main'
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  libLLVM-16.so.1 0x00007f1137fbec16 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 54
1  libLLVM-16.so.1 0x00007f1137fbcdb0 llvm::sys::RunSignalHandlers() + 80
2  libLLVM-16.so.1 0x00007f1137fbf3eb
3  libc.so.6       0x00007f1136842520
4  libc.so.6       0x00007f11368969fc pthread_kill + 300
5  libc.so.6       0x00007f1136842476 raise + 22
6  libc.so.6       0x00007f11368287f3 abort + 211
7  libLLVM-16.so.1 0x00007f1137f0ae15 llvm::report_fatal_error(llvm::Twine const&, bool) + 437
8  libLLVM-16.so.1 0x00007f1137f0ac56
9  libLLVM-16.so.1 0x00007f113a042997
10 libLLVM-16.so.1 0x00007f1138346d6b llvm::MachineFunctionPass::runOnFunction(llvm::Function&) + 635
11 libLLVM-16.so.1 0x00007f11380f4d3d llvm::FPPassManager::runOnFunction(llvm::Function&) + 653
12 libLLVM-16.so.1 0x00007f11380fa983 llvm::FPPassManager::runOnModule(llvm::Module&) + 51
13 libLLVM-16.so.1 0x00007f11380f53f5 llvm::legacy::PassManagerImpl::run(llvm::Module&) + 965
14 libLLVM-16.so.1 0x00007f113947c6ca
15 libLLVM-16.so.1 0x00007f113947d5f0
16 libLLVM-16.so.1 0x00007f113947d47f llvm::lto::thinBackend(llvm::lto::Config const&, unsigned int, std::function<llvm::Expected<std::unique_ptr<llvm::CachedFileStream, std::default_delete<llvm::CachedFileStream>>> (unsigned int, llvm::Twine const&)>, llvm::Module&, llvm::ModuleSummaryIndex const&, llvm::StringMap<std::unordered_set<unsigned long, std::hash<unsigned long>, std::equal_to<unsigned long>, std::allocator<unsigned long>>, llvm::MallocAllocator> const&, llvm::DenseMap<unsigned long, llvm::GlobalValueSummary*, llvm::DenseMapInfo<unsigned long, void>, llvm::detail::DenseMapPair<unsigned long, llvm::GlobalValueSummary*>> const&, llvm::MapVector<llvm::StringRef, llvm::BitcodeModule, llvm::DenseMap<llvm::StringRef, unsigned int, llvm::DenseMapInfo<llvm::StringRef, void>, llvm::detail::DenseMapPair<llvm::StringRef, unsigned int>>, std::vector<std::pair<llvm::StringRef, llvm::BitcodeModule>, std::allocator<std::pair<llvm::StringRef, llvm::BitcodeModule>>>>*, std::vector<unsigned char, std::allocator<unsigned char>> const&) + 2927
17 libLLVM-16.so.1 0x00007f1139476ed9
18 libLLVM-16.so.1 0x00007f1139476883
19 libLLVM-16.so.1 0x00007f113843f984
20 libLLVM-16.so.1 0x00007f1137f6bdf8 llvm::ThreadPool::processTasks(llvm::ThreadPoolTaskGroup*) + 760
21 libLLVM-16.so.1 0x00007f1137f6c893
22 libc.so.6       0x00007f1136894ac3
23 libc.so.6       0x00007f1136926660
error: failed to link /tmp/tinygo3956045343/main: signal: aborted (core dumped)
```

What's weird if I comment out a println in the new dialConnection2 func, I don't get the LLVM error:

```
+func dialConnection2() {
+       var err error
+
+       println("\r\n---------------\r\nDialing TCP connection2")
+       conn2, err = net.Dial("tcp", address2)
+       for ; err != nil; conn2, err = net.Dial("tcp", address2) {
+               //println("Connection failed:", err.Error())
+               time.Sleep(5 * time.Second)
+       }
+       println("Connected!\r")
+}
```

Now it builds fine.

aykevl, 2024-01-04:
I tried this, but:

  * On my own system (Fedora with LLVM 17) I couldn't reproduce this (`tinygo version 0.31.0-dev linux/arm64 (using go version go1.21.5 and LLVM version 17.0.6)`)
  * In a distrobox with Debian 12 and LLVM 16, I got the following error:
    ```
    $ tinygo version
    tinygo version 0.31.0-dev linux/arm64 (using go version go1.19.8 and LLVM version 16.0.6)
    
    $ tinygo build -target nano-rp2040 -stack-size 4kb -ldflags="-X 'main.ssid=test' -X 'main.pass=testtest'" ./examples/net/tlsclient/
    # net
    /home/ayke/src/tinygo/tinygo/src/net/ip.go:307:34: cannot convert p4 (variable of type IP) to [4]byte
    /home/ayke/src/tinygo/tinygo/src/net/ip.go:309:35: cannot convert ip (variable of type IP) to [16]byte
    ```

aykevl, 2024-01-06:
I can reproduce this (Debian 12, LLVM 16 from apt.llvm.org, Go 1.21.5 from https://go.dev/dl) but not in LLVM 17 (same environment but with LLVM 17). Maybe this bug got fixed in LLVM 17?

deadprogram, 2024-05-05:
@kenbell @aykevl is this still an issue or can it be closed?

aykevl, 2024-05-06:
I haven't heard anybody about this bug, and I think it's been fixed somewhere after LLVM 15. Let's close this. 
We can always reopen it if needed.

lasselj, 2024-09-17:
I am also struggling with same rather arbitrary problem. Here is a loom that shows it can be as little as a single fmt.Printf that causes the "LLVM ERROR: underestimated function size" problem: 

https://www.loom.com/share/ad62d17323d546d5864b9becb2b5f191?sid=932029c2-f387-4f78-b68d-994ec4c77ad4

aykevl, 2024-09-18:
Yeah, I'm afraid it's still present. It just happens very rarely, in specific cases. Modifying the code a bit in some arbitrary way can work around the problem.

Here is the upstream LLVM bug: https://github.com/llvm/llvm-project/issues/95683

ofauchon, 2025-04-03:
Still present on ArchLinux:

```
$ tinygo version
tinygo version 0.38.0-dev-91e20a53 linux/amd64 (using go version go1.24.1 and LLVM version 19.1.7)

$ pacman -Q | grep llvm
llvm 19.1.7-1

$ GODEBUG=gotypesalias=0  tinygo  flash -target=pico 

LLVM ERROR: underestimated function size
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.	Running pass 'Function Pass Manager' on module '/tmp/tinygo2622835140/main.o'.
1.	Running pass 'ARM constant island placement and branch shortening pass' on function '@"(*main.DigitalGauge[tinygo.org/x/drivers/pixel.RGB565BE]).run[tinygo.org/x/drivers/pixel.RGB565BE]"'
 #0 0x00007d093001a730 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /usr/src/debug/llvm/llvm-19.1.7.src/lib/Support/Unix/Signals.inc:723:22
 #1 0x00007d09300176bd llvm::sys::RunSignalHandlers() /usr/src/debug/llvm/llvm-19.1.7.src/lib/Support/Signals.cpp:105:20
 #2 0x00007d09300176bd SignalHandler /usr/src/debug/llvm/llvm-19.1.7.src/lib/Support/Unix/Signals.inc:403:31
 #3 0x00007d092f24bcd0 (/usr/lib/libc.so.6+0x3dcd0)
 #4 0x00007d092f2a5624 __pthread_kill_implementation /usr/src/debug/glibc/glibc/nptl/pthread_kill.c:44:76
 #5 0x00007d092f24bba0 raise /usr/src/debug/glibc/glibc/signal/../sysdeps/posix/raise.c:27:6
 #6 0x00007d092f233582 abort /usr/src/debug/glibc/glibc/stdlib/abort.c:81:3
 #7 0x00007d092fcda9f0 std::mutex::lock() /usr/include/c++/14.2.1/bits/std_mutex.h:117:22
 #8 0x00007d092fcda9f0 std::lock_guard<std::mutex>::lock_guard(std::mutex&) /usr/include/c++/14.2.1/bits/std_mutex.h:250:23
 #9 0x00007d092fcda9f0 llvm::install_bad_alloc_error_handler(void (*)(void*, char const*, bool), void*) /usr/src/debug/llvm/llvm-19.1.7.src/lib/Support/ErrorHandling.cpp:131:61
#10 0x00007d092ff1a1ee (/usr/lib/libLLVM.so.19.1+0x71a1ee)

```

Fixed by modifying arbitrary the code as @aykevl suggests.

aykevl, 2026-02-09:
This might have been fixed in a recent LLVM:
https://github.com/llvm/llvm-project/pull/175763

davecheney, 2026-10-26:
TinyGo 0.44 has support for LLVM 23. I tested the reproducer at https://godbolt.org/z/9cxW1Wcv8 directly with llc: it aborts with LLVM 22.1.8, but does not appear to abort with LLVM 23.1.4.

Please try a 0.44-dev build using LLVM 23, or wait for TinyGo 0.44 to ship shortly and test then. Leaving this open until we have confirmation from affected projects.