tinygo-org/tinygo #2970: On RP2040, we got incorrect sleep time after a select with timeout

elbert-wong, 2022-07-07:
In a channel select with timeout, if the select hit the non-timeout branch, we got incorrect sleep time in the following code.

we got correct sleep time if we:
comment out the timer go-routine
OR
comment out the timeout branch in select

Note: to handle the pin edge interrupt, the #2847 code is merged, the issue is reported as #2877

[image]

pinOut signal:
[image]

kenbell, 2022-07-07:
@elbert-wong - I'm not able to repro on the latest dev branch.  I'm seeing the timing below without commenting out any code (I'm using a Pico):
[image]

This is to code I'm using:
```go
package main

import (
	"machine"
	"runtime"
	"time"
)

const (
	pinIn  = machine.GPIO13
	pinOut = machine.GPIO15
	pinLED = machine.LED
)

var timeout = make(chan bool, 1)
var tstChan = make(chan bool, 1)

func pinConfig() {
	pinIn.Configure(machine.PinConfig{Mode: machine.PinInputPullup})
	pinLED.Configure(machine.PinConfig{Mode: machine.PinOutput})
	pinOut.Configure(machine.PinConfig{Mode: machine.PinOutput})
}

func main() {
	pinConfig()

	go func() {
		time.Sleep(10000 * time.Millisecond)
		timeout <- true
	}()

	pinIn.SetInterrupt(machine.PinFalling, pinFallHandler)

	select {
	case <-tstChan:
	case <-timeout:
	}

	pinIn.SetInterrupt(machine.PinFalling, nil)

	pinOut.High()
	time.Sleep(100 * time.Millisecond)
	pinOut.Low()
	time.Sleep(300 * time.Millisecond)
	pinOut.High()

	for {
		runtime.Gosched()
	}
}

func pinFallHandler(pin machine.Pin) {
	tstChan <- true
}
```

elbert-wong, 2022-07-08:
@kenbell Thank you for your prompt reply!
Could you confirm that the light sleep is enabled in your dev code? (which is also included in release 0.24.0)
If I disable the light sleep, the following sleep in main loop works well. But in that case, the select code is blocked for a very long time, if a new ext interrupt triggers again during this block time, we will get a panic (writing a full channel in interrupt). 

my issue was found on 0.22.0(with light sleep merged) and also 0.24.0

I have build a dev branch (tinygo version 0.25.0-dev linux/amd64 (using go version go1.18.3 and LLVM version 14.0.6) with outside llvm-14 this morning, still no lucky.
But I am not sure whether I am using the tinygo in dev correctly.  I just run it as below:
<path to dev >/tinygo build -target=pico -o pico.uf2 -size short github.go
And also I have copied the folder lib/compiler-rt-builtins from release 0.24.0.
If anything wrong in my steps, please let me know. Thanks!

elbert-wong, 2022-07-08:
Seems like I fixed this issue by correcting the sleep duration in addSleepTask.
But I have no idea why this code works if light-sleep is not enabled

[image]

deadprogram, 2026-10-22:
Thanks for pointing out this was fixed @elbert-wong now closing.

closed 2026-10-22 by deadprogram