tinygo-org/tinygo #2975: tinygo build fails with error: could not find build id in %!s(<nil>) on Android

wasmasmr, 2022-07-11:
I installed release v0.24.0 with `go install`. When I tried to build a simple test program with the resulting executable I got this error: `error: could not find build id in %!s(<nil>)`. Presumably it is caused by [this](https://github.com/tinygo-org/tinygo/blob/cf0b7edc78e42038a0bb522b3f1a5b76928e730e/builder/buildid.go#L95) line. My operating system is Android and my architecture is arm64. I was using [Termux](https://github.com/termux/termux-app) as my terminal. Changing the GOOS and GOARCH environment variables didn't help. `file` can read the build id of the executable. `tinygo version` outputs: `tinygo version 0.24.0 android/arm64 (using go version go1.18.3 and LLVM version 14.0.6)`.

wasmasmr, 2022-07-11:
I guess using tinygo on android wasn't as straightforward as I though. If I add `"android"` to [this](https://github.com/tinygo-org/tinygo/blob/cf0b7edc78e42038a0bb522b3f1a5b76928e730e/builder/buildid.go#L27) line the build id error goes away, but I still get different errors such as this: `/data/data/com.termux/files/usr/lib/go/src/syscall/exec_linux.go:105:13: unknown GOOS/GOARCH for syscall: android/arm64`. If I change the GOOS environment variable to linux I get different output: `error: open /data/data/com.termux/files/home/tinygo/lib/musl/arch/aarch64/bits/alltypes.h.in: no such file or directory`.

wasmasmr, 2022-07-11:
`git clone --recursive` solves the last error and rewards me with new ones: `clang-14: error no such file or directory: '/data/data/com.termux/files/home/tinygo/lib/compiler-rt-builtins/absvdi2.c'` and a couple other missing files. This was with `GOOS=linux`. Without it I still get the same errors as before. It seems that if I can find those missing files the build will probably succeed with `GOOS=linux`.

deadprogram, 2022-09-30:
Released as part of `v0.26.0`

closed 2022-09-30 by deadprogram