tinygo-org/tinygo #2984: Error handling patterns

charathram, 2022-07-13:
This is a general question and I'm posting it here because I don't see a discussion forum anywhere. Apologies in advance.

How does one handle errors from TinyGo programs? For instance, if I were writing a regular Go program, I might do something like:

```
v, err := doSomething()
if err != nil {
    fmt.Println("An error occurred")
    os.Exit(1)
}
```

Obviously, that doesn't make sense in the context of an embedded program. So, are there any typical error handling patterns that people follow when writing TinyGo programs for microcontrollers?

deadprogram, 2022-07-13:
Hello @charathram 

One pattern I have used myself, is to call a "failMessage()` function in that case, which outputs something useful via serial port for debugging.

For example, see https://github.com/tinygo-org/drivers/blob/release/examples/wifinina/mqttclient/main.go#L74

charathram, 2022-07-15:
Thank you! That's very helpful.

charathram, 2022-07-21:
It might be useful to have it documented. I can open a PR on the docs site for it - I just need to figure out the appropriate location.

davecheney, 2026-10-26:
Thanks for raising this, @charathram. A documentation PR covering embedded error-handling patterns would be very welcome. Please send one when you have a chance.