tinygo-org/tinygo #2985: "make test" fails on m1 mac with "panic: unexpected feature flags"

dkegel-fastly, 2022-07-13:
"make" succeeds, but "make test" fails with
```
=== CONT  TestBuild/Host/alias.go
panic: unexpected feature flags

goroutine 189 [running]:
github.com/tinygo-org/tinygo/compiler.(*compilerContext).isThumb(0x13300aa01?)
	/Users/dkegel/src/tinygo-org/tinygo/compiler/llvm.go:299 +0x8c
github.com/tinygo-org/tinygo/compiler.(*builder).createInvokeCheckpoint(0x14015393b80)
	/Users/dkegel/src/tinygo-org/tinygo/compiler/defer.go:144 +0x70
github.com/tinygo-org/tinygo/compiler.(*builder).createInvoke(0x1401220dba0?, {0x140045696e0?}, {0x14005a6cc20, 0x3, 0x4}, {0x0, 0x0})
	/Users/dkegel/src/tinygo-org/tinygo/compiler/calls.go:81 +0x48
github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionCall(0x14015393b80, 0x1401793c240)
	/Users/dkegel/src/tinygo-org/tinygo/compiler/compiler.go:1715 +0xae8
github.com/tinygo-org/tinygo/compiler.(*builder).createExpr(0x14015393b80, {0x105c0e150?, 0x1401793c200?})
	/Users/dkegel/src/tinygo-org/tinygo/compiler/compiler.go:1811 +0x1760
github.com/tinygo-org/tinygo/compiler.(*builder).createInstruction(0x14015393b80, {0x105c0e108?, 0x1401793c200?})
	/Users/dkegel/src/tinygo-org/tinygo/compiler/compiler.go:1323 +0x780
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0x14015393b80)
	/Users/dkegel/src/tinygo-org/tinygo/compiler/compiler.go:1222 +0x7b8
github.com/tinygo-org/tinygo/compiler.(*compilerContext).createPackage(0x1401220dba0, {0x14007930b60?}, 0x1401aa44f80)
	/Users/dkegel/src/tinygo-org/tinygo/compiler/compiler.go:831 +0x74c
github.com/tinygo-org/tinygo/compiler.CompilePackage({0x1401509deb8?, 0x14001f5fad0?}, 0x140126794a0, 0x1401aa44f80, {0x14000055418?}, 0x1045b52cd?, 0xe8?)
	/Users/dkegel/src/tinygo-org/tinygo/compiler/compiler.go:294 +0x3c8
github.com/tinygo-org/tinygo/builder.Build.func3(0x14006fbc240)
	/Users/dkegel/src/tinygo-org/tinygo/builder/build.go:354 +0x1a8
github.com/tinygo-org/tinygo/builder.runJob(0x14006fbc240, 0x14000055728?)
	/Users/dkegel/src/tinygo-org/tinygo/builder/jobs.go:222 +0x4c
created by github.com/tinygo-org/tinygo/builder.runJobs
	/Users/dkegel/src/tinygo-org/tinygo/builder/jobs.go:123 +0x44c
FAIL	github.com/tinygo-org/tinygo	1.935s
FAIL
```

aykevl, 2022-07-15:
Pretty sure the bug is here:

https://github.com/tinygo-org/tinygo/blob/b65447c7d567eea495805656f45472cc3c483e03/compiler/llvm.go#L280-L282

For arm64 Macs, the LLVM triple is `arm64-unknown-macosx10.12.0` (with `arm64`, not `aarch64` like on other OSes) so `archFamily` incorrectly returns `arm` (meaning 32-bit ARM). A simple check for `arm64` should suffice to fix this bug.

dkegel-fastly, 2022-07-15:
Thanks, that gets us to the next problem.

dkegel-fastly, 2022-07-28:
Fixed.

commit 69a6718b (2022-07-15): archFamily(): arm64 is aarch64, not arm; fixes #2985
closed 2022-07-28 by dkegel-fastly