tinygo-org/tinygo #2987: Arduino Nano 33 IoT floating pin when using interrupts

bpoetzschke, 2022-07-14:
Hi there,

I was playing around with configuring pins to input mode and came across the GPIO documentation here https://tinygo.org/docs/concepts/peripherals/gpio/.

So I configured my pin as PinInputPullup and afterwards I configured the interrupt. And in the callback of the interrup I can see the pin floating. I need to connect an external PullUp resistor to make it work. Interestingly I can also just configure the interrupt without setting any input mode before. So it seems that configuring an interrupt overrides the previous pin config.

However, the documentation of `SetInterrupts` states I need to configure the pin first.

I attached 3 examples of the behaviors I observed.

Example 1: Floating pin.
```go
func main() {
	waitSerial()

	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	input.Configure(machine.PinConfig{Mode: machine.PinInputPullup})

	input.SetInterrupt(machine.PinToggle, func(pin machine.Pin) {
		println(fmt.Sprintf("Pin toggled to %t", pin.Get()))
		led.Set(pin.Get())
	})

	for {
		time.Sleep(time.Second)
	}
```

Output:
```
21:37:17.094 -> Pin toggled to true
21:37:17.094 -> Pin toggled to false
21:37:17.094 -> Pin toggled to true
21:37:17.094 -> Pin toggled to false
21:37:17.094 -> Pin toggled to true
21:37:17.121 -> Pin toggled to false
21:37:17.121 -> Pin toggled to true
21:37:17.121 -> Pin toggled to false
21:37:17.121 -> Pin toggled to true
21:37:17.158 -> Pin toggled to false
21:37:17.158 -> Pin toggled to true
21:37:17.158 -> Pin toggled to false
21:37:17.158 -> Pin toggled to true
21:37:17.192 -> Pin toggled to false
21:37:17.192 -> Pin toggled to true
21:37:17.192 -> Pin toggled to false
21:37:17.192 -> Pin toggled to true
...
```

Example 2: Floating pin without configuring it as input at all. Produces the same output.
```go
func main() {
	waitSerial()

	led.Configure(machine.PinConfig{Mode: machine.PinOutput})

	input.SetInterrupt(machine.PinToggle, func(pin machine.Pin) {
		println(fmt.Sprintf("Pin toggled to %t", pin.Get()))
		led.Set(pin.Get())
	})

	for {
		time.Sleep(time.Second)
	}
```

Example 3: No interrupts and pin configured as PinInputPullup; works as expected
```go
func main() {
	waitSerial()

	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	input.Configure(machine.PinConfig{Mode: machine.PinInputPullup})

	for {
                println(fmt.Sprintf("Pin toggled to %t", input.Get()))
		time.Sleep(time.Second)
	}
```

Output for example 3:
```
21:40:54.231 -> Pin toggled to true
--> Button pressed
21:40:55.233 -> Pin toggled to false
21:40:56.259 -> Pin toggled to false
21:40:57.249 -> Pin toggled to false
21:40:58.254 -> Pin toggled to false
--> Button released
21:40:59.255 -> Pin toggled to true
```

jhoughjr, 2022-10-12:
I wonder if this is related to a crash I'm seeing. My pins for attachInterrupt are floating atm.
I've noticed after less than one minute the Arduino locks up completely. 

not attaching the interrupt prevents the crash, but as it worked when in situ I assume its because they are not connected to anything.
Didn't realize you could use Go on Arduino.

bpoetzschke, 2023-02-23:
The issue was going through my head tonight and I looked at it in detail a little bit.

It seems like that `setPinCfg(val uint8) {` in `machine_atsamd21g18.go` is ranging from pin numbers 0-63. However the `getPinCfg()` in the same file ranges from 0-64. This seems odd if the micronctroller has only 64 pins(?). Why would we have a 65th pin. I then found out that pin 36 is missing in the get function.

So given this knowledge and given that the input pin is number 42 I applied the following change:

```go
case 42: // PB09
		return uint8(sam.PORT.PINCFG1_8.Get()>>16) & 0xff
```

Whereas the main repository has 
```go
case 42: // PB09
		return uint8(sam.PORT.PINCFG1_8.Get()>>8) & 0xff
```

With this given patch I'm actually able to set up the interrupts without having a floating pin.
I will create a PR to address this issue.

bpoetzschke, 2023-02-23:
I create a PR and with the change from it I can run example 1 without any issues now.

deadprogram, 2023-02-27:
Reopening and marking issue with next release tag. Will close when it is actually released.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing this issue. Thanks!

closed 2023-06-14 by deadprogram