tinygo-org/tinygo #3000: wasi: imported and exported function signatures do not result in identical signatures.

james-lawrence, 2022-07-17:
I've created a standalone repository that reproduces this [issue](https://github.com/james-lawrence/tinygo-wasi-bug).

essentially functions that are imported from the runtime and functions that are exported do not generate the same function signature making them unusable.

the example repository will fail to execute with the following result (output is truncated for brevity):
```
exported github.com/james-lawrence/tinygo-wasi-bug/wasi/runtime/env.Boolean ( i32, i32, i32, i32 ) i32
imported github.com/james-lawrence/tinygo-wasi-bug/wasi/runtime/env.Boolean ( i32, i32, i32, i32, i32 ) i32
import[4] func[env.github.com/james-lawrence/tinygo-wasi-bug/wasi/runtime/env.Boolean]: signature mismatch: i32i32i32i32i32_i32 != i32i32i32i32_i32
```
reproduction steps from the repository above:
```
tinygo build -target wasi -wasm-abi generic -o module1.wasm ./impl/wasi/runtime/runtime.go
tinygo build -target wasi -wasm-abi generic -o module2.wasm main.go
go run ./cmd/wasibug/main.go
```

aykevl, 2022-07-30:
Right, I see. The way to solve this is to mark them both as "exported" - which really means to make the function available externally. For example like this:

```go
// imported function by a wasm module
//export github.com/james-lawrence/tinygo-wasi-bug/wasi/runtime/env.Boolean
func Boolean(fallback bool, keys ...string) bool
```

Functions without an `//export` pragma will use an internal calling convention that's not stable. Functions with an `//export` pragma will try to match the C calling convention.

codefromthecrypt, 2022-09-07:
@james-lawrence is there anything else to do on this one?

ack that [`//export` isn't intuitive to mark something as an import](https://github.com/tinygo-org/tinygo/pull/455#issuecomment-531603663), but for today `//export` is how it is done.

james-lawrence, 2022-09-08:
@codefromthecrypt that is very unintuitive. ;) i'll take it for a spin this weekend and see if it works. if it does I'll point out documentation that should probably be updated for others. ;)

codefromthecrypt, 2022-09-09:
@james-lawrence opened https://github.com/tinygo-org/tinygo/issues/3147 for follow-up. maybe close this out unless there's something else to do here?

james-lawrence, 2022-09-10:
Sounds good to me; will put my notes in that issue.

closed 2022-09-10 by james-lawrence