tinygo-org/tinygo #3002: fragment is larger than or outside of variable when compiling slices.Clone() generic function call

yoursunny, 2022-07-18:
**main.go**

```go
package main

// Clone is copied from golang.org/x/exp/slices package.
func Clone[S ~[]E, E any](s S) S {
        // Preserve nil in case it matters.
        if s == nil {
                return nil
        }
        return append(S([]E{}), s...)
}

func main() {
        x := make([]int, 2)
        _ = Clone(x)
}
```

Commands and error message:

```shell
$ tinygo version
tinygo version 0.24.0 linux/amd64 (using go version go1.18.4 and LLVM version 14.0.0)

$ tinygo build -target=wasm main.go
fragment is larger than or outside of variable
  call void @llvm.dbg.value(metadata i32 %s.cap, metadata !30, metadata !DIExpression(DW_OP_LLVM_fragment, 64, 32)), !dbg !37
!30 = !DILocalVariable(name: "s", arg: 1, scope: !26, file: !10, line: 4, type: !31)
error: verification error after compiling package command-line-arguments

$ ../tinygo/build/tinygo version  # also errors when testing TinyGo compiled from `fix-issue2951` branch
tinygo version 0.25.0-dev-9554cde5 linux/amd64 (using go version go1.18.4 and LLVM version 14.0.0)
```

aykevl, 2022-08-21:
Still reproducible in the dev branch.
I have simplified this to the following:

```go
package main

func Clone[S ~[]E, E any](s S) {
}

func main() {
        Clone([]int(nil))
}
```

aykevl, 2022-08-21:
The bug appears to be that `s` is somehow considered to be of type `any` (while it is of type `[]int`).

pkedy, 2022-08-21:
@aykevl Can you share the details/branch for your fix?

I run into this issue today and I can test the workaround in my use case with generics.

aykevl, 2022-08-21:
@pkedy I made a PR, see https://github.com/tinygo-org/tinygo/pull/3086

pkedy, 2022-08-22:
@aykevl I confirmed that it works for me! Thanks!

yoursunny, 2022-08-22:
I had to delete LLVM and rebuild everything from scratch, but then it is working properly in [my use case](https://github.com/usnistgov/ndn-dpdk/tree/47177c88ca89363b8fc39530290a5ad4c5784952/ndn/ndntestenv/tiny).

```shell
$ ../tinygo/build/tinygo version
tinygo version 0.26.0-dev-f6e6aca8 linux/amd64 (using go version go1.19 and LLVM version 14.0.0)
```

deadprogram, 2022-08-22:
Let's leave open until next release, when we will close it.

deadprogram, 2022-09-30:
Released as part of `v0.26.0`

closed 2022-09-30 by deadprogram