tinygo-org/tinygo #3011: Enabling feature "bulk-memory-operations"

Sunidhi-Gaonkar1, 2022-07-22:
Hi Team,
Tinygo (v0.23.0) is being used by aquasecurity/trivy (https://github.com/aquasecurity/trivy) , to run unit and integration test cases.I am working on validating Trivy on Power(ppc64le) architecture.
I was able to build Tinygo from source following the instructions mentioned here: https://github.com/tinygo-org/tinygo/blob/release/BUILDING.md
While running test cases for Trivy, unit tests are failing with the following error:

```=== RUN   TestManager_Register/happy_path
2022-07-22T01:42:07.672-0400    INFO    Loading happy.wasm...
    module_test.go:85:
                Error Trace:    /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module_test.go:85
                Error:          Received unexpected error:
                                module load error:
                                    github.com/aquasecurity/trivy/pkg/module.NewManager
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:92
                                  - module walk error:
                                    github.com/aquasecurity/trivy/pkg/module.(*Manager).loadModules
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:134
                                  - WASM module init error happy.wasm:
                                    github.com/aquasecurity/trivy/pkg/module.(*Manager).loadModules.func1
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:126
                                  - module compile error:
                                    github.com/aquasecurity/trivy/pkg/module.newWASMPlugin
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:259
                                  - invalid function[11]: memory.copy invalid as feature "bulk-memory-operations" is disabled
                Test:           TestManager_Register/happy_path
=== RUN   TestManager_Register/only_analyzer
2022-07-22T01:42:07.677-0400    INFO    Loading analyzer.wasm...
    module_test.go:85:
                Error Trace:    /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module_test.go:85
                Error:          Received unexpected error:
                                module load error:
                                    github.com/aquasecurity/trivy/pkg/module.NewManager
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:92
                                  - module walk error:
                                    github.com/aquasecurity/trivy/pkg/module.(*Manager).loadModules
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:134
                                  - WASM module init error analyzer.wasm:
                                    github.com/aquasecurity/trivy/pkg/module.(*Manager).loadModules.func1
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:126
                                  - module compile error:
                                    github.com/aquasecurity/trivy/pkg/module.newWASMPlugin
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:259
                                  - invalid function[11]: memory.copy invalid as feature "bulk-memory-operations" is disabled
                Test:           TestManager_Register/only_analyzer
=== RUN   TestManager_Register/only_post_scanner
2022-07-22T01:42:07.680-0400    INFO    Loading scanner.wasm...
    module_test.go:85:
                Error Trace:    /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module_test.go:85
                Error:          Received unexpected error:
                                module load error:
                                    github.com/aquasecurity/trivy/pkg/module.NewManager
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:92
                                  - module walk error:
                                    github.com/aquasecurity/trivy/pkg/module.(*Manager).loadModules
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:134
                                  - WASM module init error scanner.wasm:
                                    github.com/aquasecurity/trivy/pkg/module.(*Manager).loadModules.func1
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:126
                                  - module compile error:
                                    github.com/aquasecurity/trivy/pkg/module.newWASMPlugin
                                        /root/trivy/go/src/github.com/aquasecurity/trivy/pkg/module/module.go:259
                                  - invalid function[11]: memory.copy invalid as feature "bulk-memory-operations" is disabled
                Test:           TestManager_Register/only_post_scanner
=== RUN   TestManager_Register/no_module_dir
--- FAIL: TestManager_Register (0.02s)
    --- FAIL: TestManager_Register/happy_path (0.00s)
    --- FAIL: TestManager_Register/only_analyzer (0.00s)
    --- FAIL: TestManager_Register/only_post_scanner (0.01s)
```

Since Tinygo is not supported for Power, I would like to know if feature "bulk-memory-operations" gets disabled for non-supported architectures and is there a way I can enable it.
I tried to enable it using the following command `tinygo build -llvm-features "+bulk-memory"` and got the output as `error: No available targets are compatible with triple "ppc64le-unknown-linux"`

Tinygo version:tinygo version 0.23.0 linux/ppc64le (using go version go1.18 and LLVM version 14.0.0)
OS : Red Hat Enterprise Linux 8.5
Arch : ppc64le

dankegel, 2022-08-01:
Could you try the dev branch of tinygo?

Sunidhi-Gaonkar1, 2022-08-02:
This issue has been resolved by the Trivy community,they have added the code to enable "bulk-memory-operations" to their latest version(https://github.com/aquasecurity/trivy/pull/2575).
Thank you for your response!

closed 2022-08-02 by Sunidhi-Gaonkar1